alf45tar/PedalinoMini #237: [Enhancement] Midi Value Tracker

TarFilarek, 2021-06-15:
Hello!

A small enhancement idea came to my mind.

**MIDI VALUE Tracker**

[image]

_When the **VALUE** cell is left blank*, the pedal will use the last sent **VALUE**.
*it could also be a letter, eq. **T** from **TRACK** or **C** from **COPY**_

It could be used to quickly and precisely change **MIDI VALUE** sent by **ANALOG PEDAL** (potentiometer/expression pedal). It would allow control of multiple digital pedals/knobs/sliders with only one press and move of the Expression Pedal.

Right now it can be done with **BANK+**/**BANK-** (each bank with a different **MIDI VALUE** for the same **PEDAL**), but it's much less clear and comfortable.

alf45tar, 2021-06-15:
Thanks @TarFilarek for the suggestion.

I implemented the universal expression pedal in a different way. 

1. Leave empty the Send field of an analog pedal.
2. It will control the last Control Code, Note On, Note Off, Pitch Bend and Channel Pressure send by one on any other pedals.

Let me know if it meets your expectations.

closed 2021-07-28 by alf45tar

TarFilarek, 2021-09-03:
Hi!

Sorry for writing back so late. My controller was disassembled, and I couldn't check it until today.

It works absolutely brilliant! Thank You!

alf45tar, 2021-09-03:
Thanks to you @TarFilarek for your contribution