alf45tar/PedalinoMini #238: ESP32 gets very hot and WebUI crashes when opening "Options"

Sthopeless, 2021-06-20:
Hello there,

Yesterday after discovering PedalinoMini I flashed immediately into my ESP32, I believe I have followed all instructions given on the README.md (Build, Upload, Upload Filesystem Image) and I only changed the upload_port in PLatformio.ini to match my MCU.

After all is flashed I see the device reboot and shortly after I can see and connect to the AP and open the WebUI at http://192.168.4.1 but as soon I open Options tab the ESP32 crashes and reboots
I also noticed with PedalinoMini the ESP32 keeps warming up pretty heavily almost too hot to touch sometimes.

Any idea what I am doing wrong? 

Here's the serial log:
```
SYSTEM_EVENT_STA_START
Event: 1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
SYSTEM_EVENT_AP_STACONNECTED
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
Event: 18
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/mustang.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/mustang.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/mustang.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/mustang.cfg
done.
Event: 1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
SYSTEM_EVENT_AP_STACONNECTED
Event: 18
Event: 1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
SYSTEM_EVENT_AP_STACONNECTED
abort() was called at PC 0x401e0757 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4009078c:0x3ffcf910 0x40090a09:0x3ffcf930 0x401e0757:0x3ffcf950 0x401e079e:0x3ffcf970 0x401dfe77:0x3ffcf990 0x401e01ae:0x3ffcf9b0 0x401dff15:0x3ffcf9d0 0x40101182:0x3ffcf9f0 0x4012487e:0x3ffcfa30 0x400feaa2:0x3ffcfa50 0x40128d74:0x3ffcfaa0 0x40091a1a:0x3ffcfac0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.3.5-1-g85c43024c
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 260092, Internal Free Heap 230140
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 65536
PlatformIO Built Environment: esp32dev
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 16A4AE30
PSRAM not found
Bluetooth Classic disabled: 15248 bytes released
Internal Total Heap 275428, Internal Free Heap 245292
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      16A4AE30
[NVS][Global][Boot Mode]:        1
[NVS][Global][STA SSID]:         
[NVS][Global][STA Password]:     
[NVS][Global][AP SSID]:          Pedalino-16A4AE30
[NVS][Global][AP Password]:      16A4AE30
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    16A4AE30
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Ladder]:           Ladder 1 Level 352
[NVS][Global][Ladder]:           Ladder 2 Level 533
[NVS][Global][Ladder]:           Ladder 3 Level 640
[NVS][Global][Ladder]:           Ladder 4 Level 723
[NVS][Global][Ladder]:           Ladder 5 Level 806
[NVS][Global][Ladder]:           Ladder 6 Level 908

Boot NORMAL
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START
SmartConfig started
SmartConfig timeout
```

Sthopeless, 2021-06-20:
Seems to be fixed by uploading first v2.0.0 then upgraded after connecting to my WiFi and it seems to be working.
Also all the heat seems to have disappeared and feels normal again.

V1.0.0 didn't compile with some problems with one of the libraries
V2.2.1 still gave the same problems (Options tab on WebUI didn't open and ESP32 crashed and rebooted and it was extremely hot)

alf45tar, 2021-06-20:
I never seen an ESP32 going hot apart when it is charging an 18650 battery

alf45tar, 2021-06-20:
I always suggest to use the latest version. 
Did you try 2.2.2?

Sthopeless, 2021-06-20:
@alf45tar yes sorry forget to mention, my initial test was with the latest version!
I also tried more than once.
All USB powered with only a I2C Display connect.

It seems to be working alright since I upgraded from v2.0.0, I can open Options tab and the heat seems normal for a running ESP32.

My board was running Tasmota firmware for very long time and stable so I assume it's is not faulty.
I have ordered few more devkit's v1 and when they arrive I see if same happens.

Sthopeless, 2021-06-21:
I decided to Erase Flash and try again with v2.2.2 (latest)
Problem persisted, Options tab did not open and the ESP32 heats up like crazy. 
I then decided to turn off PiHole and 🤣🤣🤣.... it worked!

So the problem was on my side, still not sure why it heats up like crazy awhile PiHole is on 😄 but after WiFi is configured etc it all goes away and seems to be working OK...  

It can be closed but I have few more questions rather do it here then open another issue, if that's alright.

I am trying to configure the firmware, WiFi connection, login, theme and such.

In `Config.h` line 558 `load_factory_default` I changed host, WiFi ssid & password, AP ssid & password, http login & password and theme. 
After erasing and reflashing it connected automatically to my WiFi but all the other options were still default.

After a few more digging I found also in `Config.h` file around line 714 the `void eeprom_update_...` lines and after configure them to use the `load_factory_default` values seems to have fixed.

Wouldn't make sense to be like this by default?

Example at: https://github.com/alf45tar/PedalinoMini/blob/master/src/Config.h#L758
```
void eeprom_update_login_credentials(String username = "admin", String password = getChipId())
```
Replace with:
```
void eeprom_update_login_credentials(String username = httpUsername, String password = httpPassword)
```

Lastly, my Bluetooth device name is the `getChipId()` is it possible to change it. if yes where? 

TIA

closed 2021-06-28 by Sthopeless