alf45tar/PedalinoMini #240: 12-button plywood build with modded expression pedal

akosbeke, 2021-07-03:
Hi there,

I realized that I haven't really shared my final PedalinoMini build that I finished in January this year.

So here's a couple of details:

- it has 12 buttons (using two 6-button ladder)
- diffused LED indicators
- USB-C port
- ON/OFF switch
- 3 routed 6.3 jacks for extra controllers
- hardware MIDI ports
- the case is made of 3mm plywood so it's super lightweight but also sturdy the way I designed it

I also modded a cheap M-Audio EX-P expression pedal so it has a toe-switch. The pedal this way has two 6.3 jack plugs: one for the expression pedal and one for the toe-switch. It works absolutely amazingly.

A few photos:

[image]

[image]

[image]

[image]

[image]

jwyse, 2021-07-03:
This looks awesome. Can we see photos of the inside, and the box design?

akosbeke, 2021-07-03:
Yeah, sure.

I have these photos that were taken before I wired up the DIN sockets, so you won't see those, but other than that this is how it's constructed.

The USB-C part kinda looks horrible but I had these adapters laying around and they work. 😆 

The other interesting thing is that I had to put a mosfet before each button because they were extremely noisy. I'm pretty sure it's not the cheapest solution but I'm a software engineer not an electrical engineer and they got the job done.

I made a 3D model for the enclosure but I managed to lost it when I upgraded to a new mac...

[image]

[image]

akosbeke, 2021-07-03:
Thankfully I was wrong. I did find the 3D model.

I designed it in blender. Although I have to mention that I had to do a few tweaks here and there like chamfering the vertical posts so it won't interfere with the switches, or some holes on the back for the USB-C port, the ON/OFF switch, etc.

https://drive.google.com/file/d/1gljxhFCWMeqbo-k4A6Z_3AIgbYPNIpn8/view?usp=sharing

jwyse, 2021-07-03:
Awesome, thanks for sharing those!

DWSQUIRES, 2021-07-03:
Very cool build. Did you use got glue to diffuse the LEDs?

akosbeke, 2021-07-04:
Yep, that's what I did. I was going to frost an acrylic sheet first but then I realized that it's too meticulous and it's impossible to make a precise cut without a laser cutter which I do not have.

But after some tests it turned out that the good old-fashioned hot glue works just as well.

closed 2021-07-29 by alf45tar