alf45tar/PedalinoMini #241: Momentary3 bug / need troubleshooting help

borsei222, 2021-07-05:
Hi

[image]
This is my first PedalinoMini-Build. It's using a T-Display, two Momentary3 and one momentary 2 switch sets, Legacy MIDI out and a LM2596s-based step-down to power the ESP32 module with Pedal-standard 9V (or sometimes 18V).

It was running fine and working as expected for two weeks. After that, when powering up the Momentary3 pedals where behaving irrational, Button-presses triggered Events programmed to other buttons, other pedals and sometimes even both of that at the same time. Meanwhile the Momentary2 buttons still work as expected.
I was able to solve the issue by erasing flash and complete reprogramming but this only lasted me like one week - then the same errors came back again.

I am not sure if this is an electrical problem or if it is on the software side. Any hints on where to start troubleshooting would be appreciated.

I've read that adding capacity to the 5V rail after the step-down (and the stepdown module already carries 220uf) - should that help in my case?

Many regards
Boris

akosbeke, 2021-07-05:
HI @borsei222 

I experienced significant improvement with the added capacitor, so it definitely won't hurt.

The other thing is that these footswitches are not the most reliable things in the world. I used to have this one issue where if I didn't step quickly enough on the switch it would send false MIDI messages.
What I did to fix this was to connect a mosfet to each button. This way the footswitch triggers the mosfet if a stable voltage is present and then the mosfet itself switches the ESP module.

I'm pretty sure it's not the most economical solution since mosfests are "more expensive" (still not very expensive though 😄) but I'm a software engineer and with my limited electric knowledge this is what worked for me and works perfectly.
This issue had to do with how voltage ladders work though so I'm not sure if this is the source of your problem.

With that being said there are about a bazillion things that can go wrong and your problem could be unique but that's two possibility you can experiment with.

Nice build BTW. 😉

alf45tar, 2021-07-07:
I don't have any experience with the momentary footswitches you used. I can suggest 3DPDT footswitches connected as Latch 2.

borsei222, 2021-07-12:
Actually the footswitches are very simple NO 2-contact switches. They are wired to your schematic for Momentary-3 and as i said they were working fine for weeks. I agree that switching to momentary-2 would probably solve the issue as they seem to be resistant against the issue i am experiencing - but:

1. i'd later on probably get issues with running Expression pedals
2. I'd loose another extension channel - and i am already now a bit frustrated because Resistance-ladders did not work for me at all and i had to split 6 switches to two "channels". 

What i've found out by testing in the meantime: it seems as if the whole setup is super-sensitive on the 3.3V side, even touching any contacts on the (configured as "none" = off) extension jacks triggers switch-events. 

Should i introduce an external 3.3V power-supply or capacity to the 3.3V lane? Is there any way to control the sensitivity on the ESP32 inputs via software (yes, i can read, understand and change code)?

alf45tar, 2021-07-12:
Can you post a picture of inside? Are you using a metal case? How do you fixed the TTGo T-Display board? 
Check you power circuit. Are you using a battery? It could help for power on stability.

borsei222, 2021-07-12:
[image]
[image]
Sorry - the insides are all but tidy. Yes, it's a metal case. 
I used two-sided isolating tape (black and clear to stick it to the case.

alf45tar, 2021-07-14:
Did you added the pull-up resistors on PIN_A1 to PIN_A6?

borsei222, 2021-07-14:
Yes, they are sitting on the backside of the PCB you can see on the upper picture. I probably can draw schematics of everything as well - if needed.
Can i use PlatformIO's monitor to graph or log voltages on the ports during runtime? I suspect either crosstalk or inconsistent voltage regulation on the 3.3v side is messing up the readings on the ports and unfortunately i don't have a graphing-capable voltage meter.
Also i'm still interested in checking if altering the voltage thresholds for HI/LOW on the ports might stabilize the event triggers.

alf45tar, 2021-07-14:
What happen using the USB as power supply?

borsei222, 2021-07-14:
Same behaviour/misreadings. :(

borsei222, 2021-08-07:
Hi,
finally managed to change the pedals to momentary 2. 
Now it seems that all 2nd buttons of all pedals don't work any more and additionally the module goes totally crazy on *some* of the 1st-button presses. Could it be that my Esp32 module is broken?
Is there any way to debug the voltage readings of the module during runtime? 

Many regards,
Boris

alf45tar, 2021-08-08:
I never broken any ESP32. It is very resistant. If the reading is crazy I suggest to check the pull-up resistors and the GND wiring.
I suggest to check, re-check, re-re-check again the wiring looking also for missing wire.

closed 2021-11-09 by alf45tar