alf45tar/PedalinoMini #243: Action in the global bank mandatory

creddox, 2021-07-08:
Hi,

while building a Pedalino Mini for myself, I got stuck on triggering an action with my successfully connected button setup. Unfortunately I've decided to head straight for Bank 1 instead of the global bank to define my actions. After an extended period of checking my breadboard connections, I've started to dig through the button event handler. There I noticed that it first tries to get the actions to the first (global bank). If it got nothing back, it skipped further processing altogether at this point.
In hindsight this design does make sense (since you kind of want global actions to be defined first such as Bank+ and Bank-), however the process would have been quicker if this fact was advertised somewhere. I could not find an indicator on the actions page or in the project's README.

Possible solutions could be:

1. Change the event handler such that if it doesn't find an suitable action in the global bank, it tries to get an action from the current bank
2. Leave the event handler the way it is and change the Web UI for editing actions such that it doesn't allow editing of other banks when no action is assigned to the global bank
3. Update the README in the relevant section for the global bank

If you want help implementing any of these solutions I'd be happy to oblige and provide PRs (I've got some experience with writing code, just not as much with C).

Thanks a lot by the way for this wonderful project, I thoroughly enjoy building a specimen!

aFunkyBass, 2021-07-09:
I had the same problem with a freshly updated Pedalino Mini on my pedalboard. I had to define dummy actions on the global bank to make normal banks work.

alf45tar, 2021-07-09:
Thanks @creddox and @aFunkyBass for let me know the bug.
It is very easy to fix. I will do later today.

alf45tar, 2021-07-09:
Fixed for sponsors

closed 2021-07-09 by alf45tar