alf45tar/PedalinoMini #244: How can I move a bank?

teopost, 2021-07-17:
I would like to switch 2 banks

[image]

I tried to do this using the configuration editor, but the changes are not applied

alf45tar, 2021-07-17:
Did you saved changes using Save on the right of configuration editor?
Did you loaded the configuration?

teopost, 2021-07-17:
Yes. the configuration is saved. the json is modified but is not applied

alf45tar, 2021-07-17:
Did you changed the bank number in the Actions root of both banks?

teopost, 2021-07-18:
I didn't! Now I have succeeded
I suggest adding a more user friendly feature as it is very inconvenient to change the settings in this way.
Thanks for the support

alf45tar, 2021-07-18:
It is absolutely true.

alf45tar, 2021-07-25:
Bank order re-arrangment using drag & drop will be released with version 2.2.8.

teopost, 2021-07-25:
Wow! Fantastic

teopost, 2021-07-26:
How does it work?
Where should I drag & drop?

teopost, 2021-07-26:
Tested. It work.
Much appreciated modification
Thank you

teopost, 2021-07-26:
Ops! I found a problem!
The change is not saved

alf45tar, 2021-07-27:
Drag & drop the bank number/name.
Click on any bank to save the order into memory
Click on Save button to save into EEPROM

teopost, 2021-07-27:
It works. 
This implementation is very useful
Thank you

closed 2021-07-27 by alf45tar