alf45tar/PedalinoMini #245: response to incoming Bank/Prog Ch messages

wrpooley, 2021-07-20:
I'd like to pair my pedaling with an HX stomp but would to have some bi-directional communication between them.  Does PedalinoMini simply pass incoming midi to other interfaces or can it process them in any way? Is it possible to trigger actions based on incoming midi?  Or is the only way with a switch or pedal attached to PedalinoMini?

Regards!

alf45tar, 2021-07-21:
No, it is not possible to trigger an action based on incoming MIDI messages. Which actions do you need to trigger? 
I suppose you need a special actions (Bank+, Bank-, etc.) not a MIDI message.

wrpooley, 2021-07-21:
Yes, I think special actions are what I'm thinking.  I'm not that deep into to know how useful it would be but the idea is that Pedalino would be aware if HX Stomp changed banks or programs (with it's own switches).  That kind of behavior can be observed with the Mission Engineering controller.  Thanks for the response.

alf45tar, 2021-08-03:
When selecting different presets, HX Stomp automatically transmits a MIDI Program Change message corresponding to the selected preset. PedalinoMini will update its corresponding led on incoming Program Change (if an action exist linking the pedal, the led and the PC #).

closed 2021-11-09 by alf45tar