alf45tar/PedalinoMini #247: I want connect 12 footswitch and 1 wah pedal, but I don't understand schematic and setup via web.

e36910, 2021-07-24:
I can setup only 6 footswitch using pin 13, 12, 14, 27, 26, 25. But I don't understand how to use for exemple only pin 13 with 6 footswitch and pin 12 with 6 footswitch to get a total of 12 pedals and using pin 14 to get wah pedal. I want using photoresistor as wah pedal. How to connecting and setup this? Thank you.

alf45tar, 2021-07-25:
Ladder pedal mode requires 2 pins: one digital and one analog.

e36910, 2021-07-26:
[image]
Is it correct settings?

alf45tar, 2021-07-26:
If Pedal 1 is configured as Ladder use the Button value in every action to distinguish the 6 footswitches and assign separate actions.

e36910, 2021-07-26:
Do you have exaple sutup pedal 1 via web interface?

alf45tar, 2021-07-26:
[image]
[image]

e36910, 2021-07-26:
[image]
I use this schematic with R10. Working only 5 buttons. Schematic on the project doesn't work.

alf45tar, 2021-07-26:
Did you calibrated you ladder as indicated in the instructions?

e36910, 2021-07-26:
No, because I haven`t display. My pedal will be without display.

alf45tar, 2021-07-26:
Without a display you need to calculate the ladder levels and enter into the Options
[image]

Did you do the math?

closed 2021-09-05 by e36910