alf45tar/PedalinoMini #251: Bank 0 is always selected when changing profile (from B to A)

teopost, 2021-07-24:
My configuration has 2 profiles:
The switch between profile A and B I do it with a pedal.

* I go to profile B
* I select bank 2
* I go to profile A
* I select bank 3
* I go to profile B and the last selected bank is shown (3)
* I go back to profile A and bank 0 is selected (AND NOT BANK 2)

Is this behavior correct?

alf45tar, 2021-07-24:
Yes, it could be.
For every profile the current bank is saved with the profile. Every time a profile is selected the saved current bank is restored.
Profile changes is not designed for live performance.
Do you value more having the last bank by profile selected?

teopost, 2021-07-24:
In the profile A the bank loaded is not last saved. Is only the bank 0.

alf45tar, 2021-07-24:
Are you using the last sponsor version?

teopost, 2021-07-25:
Yes 2.2.7

alf45tar, 2021-07-25:
It is not happening in the version I am working on and not yet released.
It will be fixed in version 2.2.8.

Do you prefers to save the last bank when changing the profile?

teopost, 2021-07-25:
Yes

alf45tar, 2021-07-26:
Version 2.2.8 has been released yesterday. Check and confirm if it is ok now.

teopost, 2021-07-26:
Tested!

closed 2021-07-27 by alf45tar