alf45tar/PedalinoMini #253: Doesn't work "New Action"

e36910, 2021-07-25:
Do you have video or other manual how setup 36 CC button?

e36910, 2021-07-25:
[image]

alf45tar, 2021-07-25:
Did you “Upload File System Image” under Platform?
Do not skip this step otherwise the WebUI will not works properly.

e36910, 2021-07-25:
[image]
How is it doing?

alf45tar, 2021-07-25:
[image]

e36910, 2021-07-25:
How to setup 36 footswitch CC control? Web interface has only 7 pedal. Do you have example setup?

alf45tar, 2021-07-25:
Use the first 6 and configure as Ladder.

closed 2021-07-27 by alf45tar