alf45tar/PedalinoMini #254: How to  bluetooth connect via bias fx2 in windows10?

e36910, 2021-07-26:
Interesting driver for windows for connection bluetooth pedal with bias fx2 and other vst plugin. Do you have manual?

alf45tar, 2021-07-27:
https://www.youtube.com/watch?v=bfV4szNlFWM
https://www.youtube.com/watch?v=gYCDurXL1Qw

e36910, 2021-07-28:
Is it possible to use Wi-Fi instead of Bluetooth to transfer midi signal to Windows and ios?

alf45tar, 2021-07-28:
Yes, it is.
For iOS use https://apps.apple.com/us/app/midi-network/id1102428729
For Windows you need https://www.tobias-erichsen.de/software/rtpmidi.html

e36910, 2021-07-28:
[image]
It doesn't work

alf45tar, 2021-07-28:
Did you connected your Windows machine to the PedalinoMini hotspot?

e36910, 2021-07-28:
Yes

e36910, 2021-07-28:
Midibery  doesn't work to

alf45tar, 2021-07-29:
I replicated the issue with macOS too. It is happing when Pedalino is used as access point.
As a temporary workaround connect PedalinoMini and you PC to an access point.

e36910, 2021-07-29:
Can it be possible to connect it through bluetooth somehow? Midibery doesn't work

alf45tar, 2021-07-29:
Network MIDI/AppleMIDI/RTP MIDI fixed in the latest sponsor version

e36910, 2021-07-30:
Will the bluetooth connection work with Windows 10?
MIDI/AppleMIDI/RTP MIDI is working. Thank you!

alf45tar, 2021-07-30:
Bluetooth MIDI works for me.

The procedure is:
1. Pair PedalinoMini as any Bluetooth device in Windows 10
2. Run MIDIberry and select PedalinoMini device name in the INPUT list

If you need a MIDI port:
1. Create a MIDI port with loopMIDI
2. Select the MIDI port in the OUTPUT of MIDIberry

e36910, 2021-07-31:
I have Windows ltsc, bluetooth connection via midiberry doesn't work. I did everything according to the instructions.

alf45tar, 2021-07-31:
I do not know Windows LTSC and I do not support it because it is not for this kind of application.

closed 2021-07-31 by alf45tar