alf45tar/PedalinoMini #256: no buttons works

prostealth, 2021-07-27:
hi..

i uploaded the code successfully on doit devkit v1. UI works well. i set all the action.
but all buttons doesnt works. even the analog also doesnt work.
im sure i didnt missing any wiring. upload it again on another, same problem.

alf45tar, 2021-07-27:
It is hard to say what's wrong.
Did you tried with Pedal 7? Pedal 7 his the built-in BOOT button.
This test is useful to eliminate the wiring variable.

prostealth, 2021-07-28:
sorry, it doesnt work on pedal 7.
it seems the supplier send me the wrong board it said esp32 devkit not esp32 doit dev kit. is it the problem? should i bought the right board or i still can use this board?

alf45tar, 2021-07-28:
Esp32  boards are very similar. They works for sure with minor adjustments in Pedalino.h file.
Post a photo or the complete name or a link to better understand your model.

prostealth, 2021-07-28:
@[image]

prostealth, 2021-07-28:
[image]

prostealth, 2021-07-28:
im checking pedalino.h there's no esp32 doit dev kit only heltec kit and ttgo t display[image]

alf45tar, 2021-07-28:
Your board is a esp32 doit devkit compatible board. It should work without any changes in Pedalino.h.

The session for a doit board is here 
https://github.com/alf45tar/PedalinoMini/blob/95adcf452d1a4d9d8d7f1c82f5dc2924e6513c48/src/Pedalino.h#L75

Can you copy and paste here the PedalinoMini log?

prostealth, 2021-07-29:
> Executing task: C:\Users\Hansa\.platformio\penv\Scripts\platformio.exe run --environment esp32doit-devkit-v1 <

Processing esp32doit-devkit-v1 (board: esp32doit-devkit-v1; platform: espressif32; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.3.1) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AceButton> 1.9.0+sha.cb9c896
|-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <ESP Async WebServer> 1.2.3+sha.1d46269
|   |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.2.1+sha.216812c
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <FastLED> 3.4.0+sha.5cc17b2
|   |-- <SPI> 1.0
|-- <MD_REncoder> 1.0.1+sha.d0c998d
|-- <NimBLE-Arduino> 1.2.0+sha.0e4fa36
|-- <OSC> 0.0.0+sha.de8cf4b
|-- <ResponsiveAnalogRead> 1.2.1+sha.504c16b
|-- <TFT_eSPI> 2.3.70+sha.58f457b
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <MIDI Library> 5.0.2+sha.d501f4b
|-- <AppleMIDI> 3.2.0+sha.ee56ae8
|   |-- <MIDI Library> 5.0.2+sha.d501f4b
|-- <BLE-MIDI> 2.1.0+sha.aa0f6bc
|   |-- <MIDI Library> 5.0.2+sha.d501f4b
|   |-- <NimBLE-Arduino> 1.2.0+sha.0e4fa36
|-- <ipMIDI> 1.0.3+sha.b166047
|   |-- <MIDI Library> 5.0.2+sha.d501f4b
|-- <Preferences> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Ticker> 1.1
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
[... 315 lines cut ...]
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32doit-devkit-v1\libFrameworkArduino.a
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
Retrieving maximum program size .pio\build\esp32doit-devkit-v1\firmware.elf
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.5% (used 63748 bytes from 327680 bytes)
Flash: [========= ]  93.6% (used 1778930 bytes from 1900544 bytes)
Building .pio\build\esp32doit-devkit-v1\firmware.bin
esptool.py v3.1
Merged 1 ELF section
=========================================== [SUCCESS] Took 293.36 seconds ===========================================

Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  SUCCESS   00:04:53.361
============================================ 1 succeeded in 00:04:53.361 ============================================

Terminal will be reused by tasks, press any key to close it.

prostealth, 2021-07-29:
[image]

alf45tar, 2021-07-29:
Everything is ok.
Can you copy and paste the log during PedailnoMini booting and when you press a button?

prostealth, 2021-07-29:
where can i find the log?

alf45tar, 2021-07-29:
[image]

prostealth, 2021-07-29:
nothing happen

> Executing task: C:\Users\Hansa\.platformio\penv\Scripts\platformio.exe device monitor --environment esp32doit-devkit-v1 <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

alf45tar, 2021-07-29:
Reboot PedalinoMIni and copy and paste the log in the Monitor windows.

prostealth, 2021-07-29:
> Executing task: C:\Users\Hansa\.platformio\penv\Scripts\platformio.exe device monitor --environment esp32doit-devkit-v1 <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.3.5-1-g85c43024c
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 337164, Internal Free Heap 306444
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 65536
PlatformIO Built Environment: esp32doit-devkit-v1-wifi
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 35B267AC
PSRAM not found
Internal Total Heap 337132, Internal Free Heap 306348
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      35B267AC
[NVS][Global][Boot Mode]:        5
[NVS][Global][STA SSID]:         TP-Link_3E54
[NVS][Global][STA Password]:
[NVS][Global][AP SSID]:          Pedalino-35B267AC
[NVS][Global][AP Password]:      35B267AC
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    35B267AC
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Ladder]:           Ladder 1 Level 352
[NVS][Global][Ladder]:           Ladder 2 Level 533
[NVS][Global][Ladder]:           Ladder 3 Level 640
[NVS][Global][Ladder]:           Ladder 4 Level 723
[NVS][Global][Ladder]:           Ladder 5 Level 806
[NVS][Global][Ladder]:           Ladder 6 Level 908

Boot AP
USB MIDI started
DIN MIDI started
Skipped connection to last AP and/or SmartConfig/WPS setup
SYSTEM_EVENT_WIFI_READY
AP Pedalino-35B267AC started with password 35B267AC
AP SSID     : Pedalino-35B267AC
AP PSK      : 35B267AC
AP MAC      : AC:67:B2:35:96:95
AP IP       : 192.168.4.1
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
Channel     : 1
Connect to Pedalino-35B267AC wireless network with password 35B267AC
mDNS responder started
OTA update started
HTTP server started
Connect to http://35B267AC.local/update for firmware update
Connect to http://35B267AC.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
Internal Total Heap 332496, Internal Free Heap 204992
Loading profile ...
Reading NVS Profile Adone
Bank  2
Pedal  1     1   MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE      80   Channel  1   Pin D25   Pin A36
Pedal  2     1   ANALOG       PRESS_1       POLARITY+      Channel  0   Pin A39 D26
Pedal  3     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D27
Pedal  4     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D14
Pedal  5     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D12
Pedal  6     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D13
Pedal  7     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D0
Pedal  2   input 0 output 13 velocity 0.01
Pedal  2   input 0 output 4 velocity -55.90
Pedal  2   input 0 output 1 velocity -21.28
Pedal  2   input 0 output 0 velocity -7.63
MTC None
SYSTEM_EVENT_AP_STACONNECTED
Event: 18
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
Updating NVS Profile A ... done
Updating NVS ... done
[NVS][Global][Current Profile]: 0
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
Updating NVS Profile A ... done
Updating NVS ... done
[NVS][Global][Current Profile]: 0
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
Updating NVS Profile A ... done
Updating NVS ... done
[NVS][Global][Current Profile]: 0
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 2     ButtonState: 1
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 2     ButtonState: 1
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 2     ButtonState: 1
Pedal: 6     Button: 0    EventType: 0     ButtonState: 0
Pedal: 6     Button: 0    EventType: 1     ButtonState: 1
Pedal: 6     Button: 0    EventType: 2     ButtonState: 1

prostealth, 2021-07-29:
the button i press was boot button.
it come to pedal 6. does it supposed to be pedal 7?

alf45tar, 2021-07-29:
The internal pedal numbering is from 0 to 6 for a 7 pedals controller.
The pressing of BOOT is detected but no action is triggered. 
Can you share you actions for pedal 7 in the current profile (A) and in the current bank (2)?

prostealth, 2021-07-29:
{"BankNames":[{"Bank":0,"Name":"Global"},{"Bank":1,"Name":""},{"Bank":2,"Name":""},{"Bank":3,"Name":""},{"Bank":4,"Name":""},{"Bank":5,"Name":""},{"Bank":6,"Name":""},{"Bank":7,"Name":""},{"Bank":8,"Name":""},{"Bank":9,"Name":""},{"Bank":10,"Name":""},{"Bank":11,"Name":""},{"Bank":12,"Name":""},{"Bank":13,"Name":""},{"Bank":14,"Name":""},{"Bank":15,"Name":""},{"Bank":16,"Name":""},{"Bank":17,"Name":""},{"Bank":18,"Name":""},{"Bank":19,"Name":""},{"Bank":20,"Name":""}],"Actions":[{"Bank":1,"Pedal":1,"Button":1,"Led":1,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Event":"Press","Message":"Control Change","Channel":1,"Code":80,"Value1":127,"Value2":127},{"Bank":1,"Pedal":1,"Button":1,"Led":1,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Event":"Release","Message":"Control Change","Channel":1,"Code":80,"Value1":0,"Value2":0},{"Bank":1,"Pedal":7,"Button":1,"Led":6,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Event":"Press","Message":"Control Change","Channel":1,"Code":81,"Value1":127,"Value2":127},{"Bank":1,"Pedal":7,"Button":1,"Led":6,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Event":"Release","Message":"Control Change","Channel":1,"Code":81,"Value1":0,"Value2":0}]}

alf45tar, 2021-07-29:
You defined actions only in bank 1 but you pressed pedal 7 when current bank was 2.
Switch to bank 1 before press pedal 7.

prostealth, 2021-07-30:
no. im not using a display

prostealth, 2021-07-30:
no. im not using a display.
but does it change when i press boot button? since boot button used as changing profile.

alf45tar, 2021-07-30:
There is a bug in the public version: normal banks actions are not triggered when there are no actions in global bank.
It is already fixed in the sponsor version.
It looks like you have no actions on Global bank.

Solutions:
1. Add a dummy action to Global bank
2. Wait the next release of a public version 
3. Sponsor the project

prostealth, 2021-07-30:
thanks.. so there are bugs.. i thought something wrong with my board.. already fried one.. i'll try to take the solutions..

prostealth, 2021-08-02:
just adding dummy action to global bank.
and it works. but if there are same action in global and other bank, it fail to switch on the effects. is it bug? or are they supposed to not having same action for each bank..

alf45tar, 2021-08-02:
The same action can be repeated in every bank. Actions on bank 0 (global) are triggered before actions in current bank.
The same event can trigger as many actions you need in bank 0 or in the current bank.

closed 2021-08-05 by alf45tar