alf45tar/PedalinoMini #257: Tap tempo/midi clock sync

DWSQUIRES, 2021-07-28:
Hi there,
I would like to know how to send Midi clock signal from the pedalino to HX stomp. I have read that I need to enable midi clock master but I cant find where to do that. I have set up a tap action but the led doesn't blink and I can't detect any midi signal on my pc from the tap action.
Much thanks

alf45tar, 2021-07-28:
The tap pedal is not sending any midi message itself. It adjust the frequency of midi clock message.
Are you using a pedal for Start/Stop?

DWSQUIRES, 2021-08-05:
Does the love mode need to be working in order for the tap tempo to work?
I have defined tap event and start/MTC start but can't seem to get it to work. No rush, I haven't finished building yet.

DWSQUIRES, 2021-11-23:
Hey there, any update on the midi clock sync? I am wanting to use this feature with my hx stomp