alf45tar/PedalinoMini #260: board Lolin32lite use - need define SDA pin

RomanKlein777, 2021-08-11:
Hello, I have built hardware with and suddenly i see I can not use OLED display, because SCL - gpio 21 is not populated on that board. As gpio 23 is not used in pedalino mini, and is populated in lolin32 lite, can be used for my case :) Can you please add some define lines in DisplayOLED.h and where else needed? producing separate fiwmware would be great. I am becoming sponsor now and hoping :)

great project! 

#if defined(Lolin32lite)
#include <SSD1306Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      23
#define OLED_I2C_SCL      22
SSD1306Wire               display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif

board pinout:
https://www.mischianti.org/wp-content/uploads/2021/07/ESP32-WeMos-LOLIN32-Lite-pinout-mischianti-low.jpg

alf45tar, 2021-08-12:
If you confirm that board works I can add in the code.

RomanKlein777, 2021-08-22:
Yes it works. 

As proof of concept i did simply this 4 "pin remap" lines and pretended I have esp32doit-devkit-v1 : 

[image]

--------------------------------------------------------
#if defined(SSH1106WIRE)
#include <SH1106Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      23 //test Lolin32lite pin remap
#define OLED_I2C_SCL      22 //test Lolin32lite pin remap
SH1106Wire                display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif
#if defined(SSD1306WIRE)
#include <SSD1306Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      23 //test Lolin32lite pin remap
#define OLED_I2C_SCL      22 //test Lolin32lite pin remap
SSD1306Wire               display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif

closed 2022-07-07 by alf45tar