alf45tar/PedalinoMini #264: How to connect a distance sensor as a pedal wah?

e36910, 2021-09-05:
I have such a distance sensor and I would like to use it as a pedal wah.  Is it possible to connect it?  Is there a wiring diagram for the contacts of this sensor?
https://a.aliexpress.com/_An8FSA
[image]

alf45tar, 2021-09-06:
The sensor is not supported yet. It requires software changes too.

renne, 2021-12-11:
@alf45tar 
The driver can be adopted from [Tasmota](https://tasmota.github.io/docs/HC-SR04/).

Padding your foot on a 2 cm hole for wah/tremolo would be funny. ;-)

e36910, 2021-12-12:
https://youtu.be/mWa1He9FP5U

e36910, 2021-12-12:
https://youtu.be/wWkQJ3UaHrQ

alf45tar, 2021-12-15:
@e36910 stay tuned. I will release ultrasonic sensor support soon.

alf45tar, 2021-12-19:
@e36910 just released

closed 2021-12-20 by alf45tar