alf45tar/PedalinoMini #265: Do you have a project for esp8266 ?

e36910, 2021-09-05:
Do you have the same project Pedalino Mini for esp8266 ?

alf45tar, 2021-09-06:
Make no sense to port PedalinoMini to esp8266.
esp32 board are cheap enough and esp8266 spec are not enough.

e36910, 2021-09-06:
I meant the NodeMcu V3 CH340 board which costs $ 2.
I could not find an esp32 board cheaper than $ 4

https://aliexpress.ru/item/32665100123.html?_t=gps-id%3AaerPdpSubstituteRcmd%2Cscm-url%3A1007.33958.210224.0%2Cpvid%3A2bbe1c3d-b3d1-45b7-91d7-d7b779d3ca72%2Ctpp_buckets%3A21387%230%23233228%23100&gps-id=aerPdpSubstituteRcmd&item_id=32665100123&pvid=2bbe1c3d-b3d1-45b7-91d7-d7b779d3ca72&scm=1007.33958.210224.0&scm-url=1007.33958.210224.0&scm_id=1007.33958.210224.0&sku_id=66676497214&spm=a2g2w.detail.1000060.1.69b41f3e30pnVg

alf45tar, 2021-09-06:
4$ is cheap enough for me

e36910, 2021-09-06:
Ok )

e36910, 2021-09-06:
When I ordered esp32 for the first time, the Chinese gave me esp8266 v3 instead. I have it idle, I thought maybe you have the same project for it )

closed 2021-09-10 by alf45tar