alf45tar/PedalinoMini #269: Which environment should I select for ESP32

NeidalMcCanuk, 2021-09-20:
Hello and good night.

I have a question about which environment select in the "Inspect" tab of Platformio (inside VSCode). I want to purchase a ESP32 but, it have 38 pins and is named as "Esp32 38 Pines Esp Wroom 32" (https://articulo.mercadolibre.com.mx/MLM-727796906-esp32-38-pines-esp-wroom-32-_JM).

After browse and select the project, it lists (the ESP32):
esp32dev
esp32doit-devkit-v1
esp32doit-devkit-v1-ble
esp32doit-devkit-v1-wifi

Then, should i use "esp32dev"?

Thanks and excuse my simple question.

alf45tar, 2021-09-21:
esp32dev or esp32doit-devkit-v1 is usually works pretty well

DWSQUIRES, 2021-09-28:
I have successfully used esp32dev for a 38 pin board

closed 2021-12-11 by alf45tar