alf45tar/PedalinoMini #270: mounting SPIFFS fails for Lolin D32 Pro

michaelhauser, 2021-09-21:
I'm trying to use a Wemos Lolin D32 Pro that I had sitting around. Terminal output includes the following:

> PSRAM found
> Bluetooth Classic disabled: 15348 bytes released
> Internal Total Heap 256388, Internal Free Heap 225832
> PSRAM Total Heap 4194252, PSRAM Free Heap 4194252
> E (1091) SPIFFS: mount failed, -10025
> [E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
> 'nvs' partition was successfully initialized

and attempted writes to the filesystem return:

> [E][vfs_api.cpp:22] open(): File system is not mounted

Seems like I should try different partition sizes. Any suggestions?

alf45tar, 2021-09-21:
I do not think is a partition size issue. Which is image did you upload?

michaelhauser, 2021-09-21:
I did Upload All.
Same error after second boot.

michaelhauser, 2021-09-21:
rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.3.5-1-g85c43024c
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 241032, Internal Free Heap 210676
PSRAM Total Heap 4194252, PSRAM Free Heap 4194252
nvs Total 65536
PlatformIO Built Environment: lolin_d32_pro
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: A42DE6B4
PSRAM found
Bluetooth Classic disabled: 15348 bytes released
Internal Total Heap 256388, Internal Free Heap 225928
PSRAM Total Heap 4194252, PSRAM Free Heap 4194252
E (1094) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
'nvs' partition was successfully initialized
Reading NVS Global ... done

alf45tar, 2021-09-22:
Update All is not the correct procedure.

The correct one is:
1. Select one board (for example esp32doit-devkit-v1)
2. Erase Flash
3. Upload
4. Upload System Image

[image]

michaelhauser, 2021-09-22:
That worked. Thanks. Time for me to RTFM on platformio.

closed 2021-09-22 by alf45tar