alf45tar/PedalinoMini #271: How do I transfer my settings from one board to another?

e36910, 2021-09-24:
I would like to save my settings from one board and transfer them to another. How can you do this correctly?
I do everything as in the picture, but nothing works.
[image]

alf45tar, 2021-09-24:
On source board

1. Save the current configuration with a anme using New button
2. Select the just saved configuration from Available Configurations and click Download button

On destination board

1. Click on Choose file under Upload Configuration
2. Select the file you just downloaded from source board and click Upload
3. Select the configuration just uploaded from Available Configuration and press Apply or Apply & Save
4.

e36910, 2021-09-24:
I did so, but nothing worked.

alf45tar, 2021-09-25:
I need more information about error or at least the log to serial port.

closed 2021-10-31 by alf45tar