alf45tar/PedalinoMini #275: Interference when connecting to PC via Bluetooth or wifi

DWSQUIRES, 2021-09-28:
When I connect pedalinomini to my pc via Bluetooth or wifi using rptmidi pedalinomini starts sending many midi messages without any pedal pushes. I currently have it all laid out on a breadboard but I have removed all wires except for power to esp32 (via 3.3v breadboard supply) and connection to the oled display and still have this problem. 
If Bluetooth on the pc is switched off and rtpmidi connection disabled then it works fine. 
I can connect to configuration pages via my phone or pc without this issue.
Has anyone else resolved this issue?

alf45tar, 2021-09-29:
This is a common behaviour when there is a pedal setup as Analog and there is no pull up resistor.

DWSQUIRES, 2021-09-29:
I should have added that I have one momentary pedal with pull up to analogue and all others disabled in the configuration

DWSQUIRES, 2021-09-30:
I've found the issue. If I touch the ws2812 strip the behaviour stops, let go of it and it starts. Specifically if I have a wire off pin D5 and touch it the crazy midi stops, I let go and it starts. Is it possible that the esp32 receives midi data through pin D5? If so, could we disable this?

alf45tar, 2021-09-30:
D5 is an output pin to drive the led strip. The electromagnetic disturbance can coming only from an input pin.
The behaviour is typical when there is something floating.
I can suggest the following:
1. Check if the led strip and the esp32 are sharing the GND
2. Disable the input from legacy MIDI interface is you are not using it

closed 2021-10-16 by DWSQUIRES