alf45tar/PedalinoMini #277: pedal 1 button 2 not working

prostealth, 2021-10-02:
i just uploaded public build on esp32 doit devkit v1
the button 2 on pedal not working
here is the log
Loading profile ...
Reading NVS Profile A ... done
Bank  2
Pedal  1     1   MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE      80   Channel  1   Pin D25   Pin A36
Pedal  2     1   MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE      82   Channel  1   Pin D26   Pin A39      
Pedal  3     1   MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE      76   Channel  1   Pin D27   Pin A34
Pedal  4     1   MOMENTARY2   PRESS_1       POLARITY+   CONTROL_CHANGE      78   Channel  1   Pin D14   Pin A35
Pedal  5     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D12
Pedal  6     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D13
Pedal  7     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D0
MTC None
Pedal: 1     Button: 1    EventType: 0     ButtonState: 0
Action:
CONTROL CHANGE.....Code  80......Value 127.....Channel  1
Pedal: 1     Button: 1    EventType: 1     ButtonState: 1
Action:
CONTROL CHANGE.....Code  80......Value   0.....Channel  1
Pedal: 1     Button: 1    EventType: 2     ButtonState: 1
SYSTEM_EVENT_AP_STACONNECTED
dhcps: send_nak>>udp_sendto result 0
Event: 18
AppleMIDI Connected Session ID: 25404865 Name: Hansa-PC
Pedal: 1     Button: 1    EventType: 0     ButtonState: 0
Action:
CONTROL CHANGE.....Code  80......Value 127.....Channel  1
Pedal: 1     Button: 1    EventType: 1     ButtonState: 1
Action:
CONTROL CHANGE.....Code  80......Value   0.....Channel  1
Pedal: 1     Button: 1    EventType: 2     ButtonState: 1

the switch is ok when im checking using multimeter
i try to upload on another board but the problem still persist.

alf45tar, 2021-10-02:
Did you connected button 2 to pin 39?

prostealth, 2021-10-03:
i connect it to vp pin..
i dont know where is pin 39.

prostealth, 2021-10-03:
why battery and A1 have same pin?
is it okay?
[image]

prostealth, 2021-10-03:
A2 to A4 are working well.
i didnt test A5 and A6 since they're  TRS jacks.
only button 2 / A1 that doesnt work

alf45tar, 2021-10-03:
Sorry I was wrong. A1 is pin 36 or VP.
There is not enough pin for battery on some board. I think you identified the issue. I suggest to delete or comment the following line

https://github.com/alf45tar/PedalinoMini/blob/93ae329235076b2dc7be8abaa3794b6e8d870a12/platformio.ini#L11

or assign BATTERY_PIN to an unused pin.

prostealth, 2021-10-04:
yes.. now all is well..

closed 2021-10-04 by alf45tar