alf45tar/PedalinoMini #280: Compatibility with ESP32-S2? Unable to build

seafoams, 2021-10-05:
Since the ESP32-S2 has integrated USB, I figured it'd be a nice all-in-one solution instead of buying two separate boards to have USB MIDI functionality. The ESP32-S2 is supported by Arduino core for the ESP32. Unfortunately, I'm having trouble with building in PlatformIO. I'm using the Adafruit Metro ESP32-S2. 

I've changed platformio.ini to include:  

    platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2  
    platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git  
    framework = arduino  
    board = metroesp32-s2  
    board_build.mcu = esp32s2 			; change microcontroller  
    board_build.f_cpu = 240000000L		; change MCU frequency   
as is recommended in PlatformIO's Espressif documentation [here](https://docs.platformio.org/en/latest/boards/espressif32/featheresp32-s2.html) and [here](https://github.com/platformio/platform-espressif32/pull/483).

Since the ESP32-S2 does not have Bluetooth, I've disabled all BLE related inclusions and flags in platformio.ini and PedalinoMini.cpp. However, I'm still getting a mountain of errors. 

Is there simply no easy way to build for the S2 without a major rework? I've been starting to go through and disable everything else non-essential, but this seems to be way above my skill level. I've attached the relevant files and diagnostics for where I'm currently at. Would greatly appreciate any advice!

[ESP32-S2 attempt.zip](https://github.com/alf45tar/PedalinoMini/files/7289709/ESP32-S2.attempt.zip)

alf45tar, 2021-10-06:
I do not see the advantages to use an ESP32-S2 but I can be wrong. 

Cons
1. one core
2. no bluetooth
3. more expensive

Pro
1. USB host but PedalinoMini is born to be wireless.

seafoams, 2021-10-07:
Understood, though I only needed USB hosting since this'll be powered directly by my PC. PedalinoMini is just far more robust and fully-featured than other midi controllers I've found on github. I appreciate all the work you've done for it! I'll most likely be getting an ESP32 to program this more easiy.