alf45tar/PedalinoMini #292: Feature request: Flash ESP32 via Chrome Web Serial API

renne, 2021-12-04:
Hi,

currently OS-specific command-line tools or IDEs are necessary to flash PedalinoMini on the ESP32.
This makes the use of PedalinoMini impossible for non IT-savy musicians.

I suggest flashing the ESP32 via the [Chrome Web Serial API](https://codelabs.developers.google.com/codelabs/web-serial#0).
[Tasmota](https://tasmota.github.io/) uses the Chrome Web Serial API [here](https://tasmota.github.io/install/).

alf45tar, 2021-12-04:
It looks interesting. Thanks @renne to let me know that technology. I will use for sure for flash PedalinoMini.

mknerr, 2021-12-04:
Thank you! I just encountered this as well while working with a smart home project (ESPhome), and was blown away. I have been able to get platformIO working most of the time, but it has definitely been a source of frustration at times, and an obstacle I

alf45tar, 2021-12-11:
Stay tuned. It will be released soon.

alf45tar, 2021-12-11:
Just released

mknerr, 2021-12-11:
Amazing. Can't wait to try it.

renne, 2021-12-12:
@alf45tar 
I just installed PedalinoMini on a TTGO T-Display via Chrome (Version 96.0.4664.93 (Offizieller Build) (64-Bit)) on Ubuntu 21.04.

Great!

closed 2021-12-12 by alf45tar