alf45tar/PedalinoMini #295: Schematics for the sponsors case version

gustavsl, 2021-12-07:
I've been tinkering with the many possibilities of the firmware and found out that the 8 button configuration from the sponsor's case which I 3D printed would suit me well.
Is there any schematics or further details to build such a setup?
Thanks a lot for your work!

jisv48, 2021-12-07:
Hola Gustavo,

What type of material did you use on the printing of the case?  I appreciate your assistance since this will be the first time I’ll 3D print something. Thanks Jose Salazar

gustavsl, 2021-12-07:
Hey @jisv48! I printed this in red PLA, 0.25 layer height and 20% triangle infill. Feels solid enough.

alf45tar, 2021-12-07:
@gustavsl added schematics here https://github.com/alf45tar/PedalinoMini-Case-1
PedalinoMini Case 1 in PLA is indestructible. Charge with you weight (before to add the foot switches of course) and let me know if you broken it. I don't think so.

gustavsl, 2021-12-07:
@alf45tar thank you so much for this! All the switches are SPDT, right?

And you were right, attached is a pic of me standing over the case :) Sturdy as hell.
[image]

alf45tar, 2021-12-07:
I use 3PDT because they are generally available.
SPDT are fine too.

closed 2021-12-07 by gustavsl