alf45tar/PedalinoMini #299: ttgo cannot be built

bobvc133, 2021-12-19:
Executing task: C:\Users\dada\.platformio\penv\Scripts\platformio.exe run --environment ttgo-t-display <

Processing ttgo-t-display (platform: espressif32; framework: arduino; board: esp32dev)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AceButton> 1.9.1+sha.5a11b88
|-- <ArduinoJson> 6.18.5+sha.ed342b9
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <ESP Async WebServer> 1.2.3+sha.1d46269
|   |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5+sha.ed342b9
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.2.1+sha.216812c
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <FastLED> 3.4.0+sha.1d236ab
|   |-- <SPI> 1.0
|-- <Improv> 1.0.0+sha.e8d5177
|-- <MD_REncoder> 1.0.1+sha.d0c998d
|-- <NimBLE-Arduino> 1.3.1+sha.4a6111d
|-- <OSC> 0.0.0+sha.de8cf4b
|-- <ResponsiveAnalogRead> 1.2.1+sha.504c16b
|-- <TFT_eSPI> 2.4.0+sha.eee56b2
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <MIDI Library> 5.0.2+sha.8f8c7cf
|-- <AppleMIDI> 3.1.2+sha.df105dc
|   |-- <MIDI Library> 5.0.2+sha.8f8c7cf
|-- <BLE-MIDI> 2.1.1+sha.5c42e14
|   |-- <MIDI Library> 5.0.2+sha.8f8c7cf
|   |-- <NimBLE-Arduino> 1.3.1+sha.4a6111d
|-- <ipMIDI> 1.0.3+sha.b166047
|   |-- <MIDI Library> 5.0.2+sha.8f8c7cf
|-- <Preferences> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Ticker> 1.1
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
[... 82 lines cut ...]
           ^
.pio\libdeps\ttgo-t-display\TFT_eSPI\Extensions/Sprite.cpp:2208:11: warning: unused variable 'pX' [-Wunused-variable]
   int32_t pX      = 0;
           ^
.pio\libdeps\ttgo-t-display\TFT_eSPI\Extensions/Sprite.cpp:2209:11: warning: unused variable 'pY' [-Wunused-variable]
   int32_t pY      = y;
           ^
.pio\libdeps\ttgo-t-display\TFT_eSPI\Extensions/Sprite.cpp:2210:11: warning: unused variable 'line' [-Wunused-variable]
   uint8_t line = 0;
           ^
.pio\libdeps\ttgo-t-display\TFT_eSPI\Extensions/Sprite.cpp:2211:8: warning: unused variable 'clip' [-Wunused-variable]
   bool clip = xd < _vpX || xd + width  * textsize >= _vpW || yd < _vpY || yd + height * textsize >= _vpH;
        ^
Compiling .pio\build\ttgo-t-display\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\ttgo-t-display\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\ttgo-t-display\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\ttgo-t-display\FrameworkArduino\esp32-hal-ledc.c.o
*** [.pio\build\ttgo-t-display\libf3f\TFT_eSPI\TFT_eSPI.cpp.o] Error 1
============================================================================================================================================= [FAILED] Took 11.50 seconds =============================================================================================================================================
Environment     Status    Duration
--------------  --------  ------------
ttgo-t-display  FAILED    00:00:11.502
======================================================================================================================================== 1 failed, 0 succeeded in 00:00:11.502 ========================================================================================================================================终端进程“C:\Users\dada\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'ttgo-t-display'”已终止，退出代码: 1。

终端将被任务重用，按任意键关闭。

alf45tar, 2021-12-19:
Did you try a Clean All before Build?

alf45tar, 2021-12-19:
The latest version of TFT_eSPI library has an issue. To use the previous one edit platform.ini and modify lib_deps with

`https://github.com/Bodmer/TFT_eSPI#20ad7e2`

alf45tar, 2021-12-19:
Temporary fix non anymore need. TFT_eSPI library fixed.

closed 2021-12-19 by alf45tar

bobvc133, 2021-12-19:
received, thank you