alf45tar/PedalinoMini #303: I have problem with guitar rig 6

e36910, 2021-12-21:
I heed setup device for guitar rig 6, I choose PRESS, Control Change, 0 - 127, but command working only 127.F.E.:My pedal only ON, when I pressed 2nd time my pedal doesn't OFF.  How is it fix?

alf45tar, 2021-12-22:
Try to add a RELEASE action. If you are using a latch switch the first press is a PRESS event and the second is a RELEASE event.

e36910, 2021-12-23:
There is a conflict if you assign CC (channel1) and PC(channel2) to one button.

alf45tar, 2022-01-01:
What do you mean with conflict? It is possible to assign multiple MIDI messages to the same action on a pedal. When the action is triggered all the messages will be sent.

e36910, 2022-01-10:
When I prescribe to 1 button СС(ch1) and PC(ch2) guitar rig 6 does not correctly perceive commands CC. As soon as I delete the PC of the guitars the rig works correctly with CC. Another words: СС(ch1) and PC(ch2) = 127,127,127....  only CC(ch1) = 0,127,0,127...

alf45tar, 2022-01-14:
I replicated your issue and it is not a bug but a limit. When you have multiple actions you cannot use the "invert previous state" in CC messages.

However to accomplish your requirements use the following setup. Set Value 1 and Value 2 to a fixed value. One for Press and one for Release.

[image]

alf45tar, 2022-01-22:
I am closing the issue. Reopen it if the solution provided do not works for you.