alf45tar/PedalinoMini #304: Legacy MIDI input doesnt work

prostealth, 2021-12-22:
hi, im installing pedalino on ttgo t-display.
all interface are works except midi input on my Din 5 pin connection. how can i know if midi input working? will it shown on the lcd display?

prostealth, 2021-12-22:
i added a led and it blinking when i press my keyboard. but still no output either on DIN, BLE, or wifi. i use a 1n4148 diode since i dont have 4001 diode..

alf45tar, 2021-12-22:
By default MIDI messages are shown on the display. Routing of incoming messages to other interfaces must be enabled in Interface webui page.

prostealth, 2021-12-22:
i enable it all but the problem still persist

prostealth, 2021-12-22:
it can receive midi messages from wifi or ble. and yes now i know it shown. it just not work on MIDI DIN 5 pin.

prostealth, 2021-12-22:
as i look to this code
#elif defined TTGO_T_DISPLAY
#undef  PEDALS
#define PEDALS                8
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_17, GPIO_NUM_35, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_37, GPIO_NUM_38, GPIO_NUM_39, GPIO_NUM_32, GPIO_NUM_33, GPIO_NUM_35, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button 2
#define USB_MIDI_IN_PIN       GPIO_NUM_21   // SDA
#define USB_MIDI_OUT_PIN      GPIO_NUM_22   // SCL
#define DIN_MIDI_IN_PIN       GPIO_NUM_15
#define DIN_MIDI_OUT_PIN      GPIO_NUM_2
#define BATTERY_PIN           GPIO_NUM_34   // Pin connected to VBAT
#define BATTERY_ADC_EN        GPIO_NUM_14   // ADC_EN is the ADC detection enable port
#define FASTLEDS_DATA_PIN     GPIO_NUM_15

the DIN MIDI IN is set to GPIO_NUM_2. but in the schematic it set to GPIO_NUM_21 while the pin is defined as USB_MIDI_IN_PIN.
should i connect it to GPIO_NUM_2 or i need to change some code?

prostealth, 2021-12-23:
as i look to this code
#elif defined TTGO_T_DISPLAY
#undef PEDALS
#define PEDALS 8
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_17, GPIO_NUM_35, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_37, GPIO_NUM_38, GPIO_NUM_39, GPIO_NUM_32, GPIO_NUM_33, GPIO_NUM_35, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN GPIO_NUM_0 // Button 2
#define USB_MIDI_IN_PIN GPIO_NUM_21 // SDA
#define USB_MIDI_OUT_PIN GPIO_NUM_22 // SCL
#define DIN_MIDI_IN_PIN GPIO_NUM_15
#define DIN_MIDI_OUT_PIN GPIO_NUM_2
#define BATTERY_PIN GPIO_NUM_34 // Pin connected to VBAT
#define BATTERY_ADC_EN GPIO_NUM_14 // ADC_EN is the ADC detection enable port
#define FASTLEDS_DATA_PIN GPIO_NUM_15

the DIN MIDI IN is set to GPIO_NUM_15. but in the schematic it set to GPIO_NUM_21 while the pin is defined as FASTLEDS_DATA_PIN.
should i connect it to GPIO_NUM_15or i need to change some code?

alf45tar, 2021-12-23:
You are right. There are some confusion between code and schematic.
The reason is that there are not enough pins to support everything. It is up to you to decide what to implement and what not.
If you do not need leds you can use pin 15 and 2. If you do not need USB midi you can use pin 21 and 22.

prostealth, 2021-12-26:
edited the code and it workd well

closed 2021-12-26 by prostealth

Wapata, 2023-06-25:
> You are right. There are some confusion between code and schematic. The reason is that there are not enough pins to support everything. It is up to you to decide what to implement and what not. If you do not need leds you can use pin 15 and 2. If you do not need USB midi you can use pin 21 and 22.

Hi, i plan to use lader with A3 and A4. Can it free up D3 and D4 for me to use legacy Midi without destroying everything ?
(And use rx and tx for USB midi and data for led...)

alf45tar, 2023-06-25:
Define your PIN assignment in Pedalino.h and try if there is no conflict.