alf45tar/PedalinoMini #306: Can pedalino connecting to bluetooth device like guitar pedal?

prostealth, 2021-12-26:
hi, i have problem connecting pedalino to a guitar pedal. since both pedalino and the pedal dont have a menu for connecting bluetooth. is there a way to connect them?

alf45tar, 2021-12-26:
Short answer: It is not possible because both are peripheral devices. At least one central device is need.

Long answer: Bluetooth LE devices are separated into two classes: central devices and peripheral devices. Peripheral devices, like a fitness device, smart tag, or MIDI controller, offer data services which can be read by other devices. Central devices, like a laptop computer or mobile phone or tablet, subscribe to and read services from peripheral devices. When a central device connects to a peripheral, it browses the available services and can then read, write, or subscribe to changes in the service.

closed 2022-01-01 by alf45tar

prostealth, 2022-01-15:
i use 2 peripherals and 1 central device (android smartphone) to route the midi and it works