alf45tar/PedalinoMini #310: what is the maximum number of Leds?

prostealth, 2022-01-18:
does 10 is maximum number of the leds?
and what is the meaning of slot in the web UI menu?

TarFilarek, 2022-01-21:
Hi!

> does 10 is maximum number of the leds?

No. You can change the number of LEDs in the code.

src > Pedalino.h > Line 27 "#define LEDS"

```
#define LEDS             10   // number of WS2812B leds
```

Change "10" to your desired number of LEDs. I'm not sure if there is any limit other than the physical limitations due to LED's power consumption. In my build, I'm using 17 LEDs and it works without any problems.

> what is the meaning of slot in the web UI menu?

I've once asked the same question. Here is the **alf45tar**'s previous response.

> SLOT is not used at the moment. In my mind it should indicate a slot on display to show the pedal tags.

alf45tar, 2022-01-22:
Thanks @TarFilarek your answers are perfect.
Memory and power consumption must be considered but for a real PedalinoMini application both are not a real issue.

@prostealth how long is your led string?

prostealth, 2022-01-22:
Thanks @TarFilarek thats really helpful
@alf45tar have 8 on my pedal but i add a socket just in case i need more leds..

closed 2022-01-22 by alf45tar