alf45tar/PedalinoMini #312: Web interface "options" tab stuck

jsleroy, 2022-01-23:
Hi,
When I select the "options" tab in the web interface, nothing is displayed. Trying to find the cause of that, it seems that it is in `WebConfigAsync.h`, call to `netword = WiFi.scanNetworkds` never returns.

alf45tar, 2022-01-23:
It never happens to me. `WiFi.scanNetworks` usually takes some seconds but it always ends.
Which ESP32 board are you using?

jsleroy, 2022-01-23:
I'm using a esp32doit-devkit-v1 board. I've the same behavior with another TTgo board.

jsleroy, 2022-01-24:
Let me know if you need some experiments on my side in order to help you to debug/fix that.

alf45tar, 2022-01-24:
Can you provide the output log?
Did you recompiled without WiFi.scanNetworks?

jsleroy, 2022-01-24:
How do you enable output log ?
Yes, recompiled without the call to scanNetworks, and it works

alf45tar, 2022-01-24:
Log is enabled by default. Just open the Monitor in Visual Studio Code.

jsleroy, 2022-01-24:
I'm not using Visual Studio Code but platformio in command line. Will check how to monitor the output and get back to you.

jsleroy, 2022-01-24:
This is the output log:
```
WiFi Provisioning timeout
SmartConfig started
SmartConfig timeout
WPS started
WPS timeout
AP Pedalino-45286F24 started with password 45286F24
AP SSID     : Pedalino-45286F24
AP PSK      : 45286F24
SYSTEM_EVENT_AP_START
AP MAC      : 24:6F:28:45:12:19
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
AP IP       : 192.168.4.1
Channel     : 1
Connect to Pedalino-45286F24 wireless network with password 45286F24
mDNS responder started
OTA update started
HTTP server started
Connect to http://45286F24.local/update for firmware update
Connect to http://45286F24.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
Internal Total Heap 268680, Internal Free Heap 92072
Loading profile ...
Reading NVS Profile A ... [E][Preferences.cpp:504] getBytes(): not enough space in buffer: 408 < 476
done
Bank  2
Pedal  1     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D25
Pedal  2     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D26
Pedal  3     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D27
Pedal  4     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D14
Pedal  5     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D12
Pedal  6     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D13
MTC None
SYSTEM_EVENT_AP_STACONNECTED
Event: 18
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
SYSTEM_EVENT_STA_START
Event: 1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
SYSTEM_EVENT_AP_STACONNECTED
SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
```

alf45tar, 2022-01-24:
I think the issue is

`Reading NVS Profile A ... [E][Preferences.cpp:504] getBytes(): not enough space in buffer: 408 < 476`

Did you modified any dimensions at the top of `Pedalino.h`? There is not enough space on EEPROM. Partitions need to be modified.

jsleroy, 2022-01-24:
No, only this change:
```
-#define LEDS             10   // number of WS2812B leds
+#define LEDS              3   // number of WS2812B leds
```
I will rerun with the call to scanNetworks commented and get the output log.

jsleroy, 2022-01-24:
Log without call to scanNetworks:
```
WiFi Provisioning timeout
SmartConfig started
SmartConfig timeout
WPS started
WPS timeout
AP Pedalino-45286F24 started with password 45286F24
AP SSID     : Pedalino-45286F24
AP PSK      : 45286F24
SYSTEM_EVENT_AP_START
AP MAC      : 24:6F:28:45:12:19
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
AP IP       : 192.168.4.1
Channel     : 1
Connect to Pedalino-45286F24 wireless network with password 45286F24
mDNS responder started
OTA update started
HTTP server started
Connect to http://45286F24.local/update for firmware update
Connect to http://45286F24.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
Internal Total Heap 268676, Internal Free Heap 91968
Loading profile ...
Reading NVS Profile A ... [E][Preferences.cpp:504] getBytes(): not enough space in buffer: 408 < 476
done
Bank  2
Pedal  1     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D25
Pedal  2     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D26
Pedal  3     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D27
Pedal  4     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D14
Pedal  5     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D12
Pedal  6     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D13
MTC None
SYSTEM_EVENT_AP_STACONNECTED
Event: 18
Event: 18
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
```
The NVS error `Reading NVS Profile A ... [E][Preferences.cpp:504] getBytes(): not enough space in buffer: 408 < 476` is still in the log, but the option page is correctly displayed now.

alf45tar, 2022-01-25:
I think you created the EEPROM when LEDS was 10 and the space occupied is 476. Now you changed LEDS to 3 and it is expecting to read only 408 bytes but the object save into EEPROM is 476.
Reset PedalinoMini to factory default after recompilation with a different LEDS value.

jsleroy, 2022-01-26:
I've reset to factory default, and the error is gone. But the options page is still not displayed and seems stuck.

See log here:
```
Boot NORMAL
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
SYSTEM_EVENT_WIFI_READY
WiFi provisioning started
SYSTEM_EVENT_STA_START
WiFi Provisioning timeout
SmartConfig started
SmartConfig timeout
WPS started
WPS timeout
AP Pedalino-45286F24 started with password 45286F24
AP SSID     : Pedalino-45286F24
AP PSK      : 45286F24
SYSTEM_EVENT_AP_START
AP MAC      : 
AP IP       : 
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
Channel     : 1
Connect to Pedalino-45286F24 wireless network with password 45286F24
mDNS responder started
OTA update started
HTTP server started
Connect to http://45286F24.local/update for firmware update
Connect to http://45286F24.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
Internal Total Heap 268640, Internal Free Heap 91936
Loading profile ...
Reading NVS Profile A ... done
Bank  2
Pedal  1     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D25
Pedal  2     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D26
Pedal  3     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D27
Pedal  4     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D14
Pedal  5     1   MOMENTARY1   PRESS_1       POLARITY+      Channel  0   Pin D12
Pedal  6     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D13
MTC None
SYSTEM_EVENT_AP_STACONNECTED
Event: 18
SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
SYSTEM_EVENT_STA_START
Event: 1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
SYSTEM_EVENT_AP_STADISCONNECTED
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
SYSTEM_EVENT_AP_STACONNECTED
```
Once again, the option page is displayed correctly and not stuck if I comment the call to `WiFi.scanNetworks()`.

jsleroy, 2022-01-31:
I commented the call to `scan.Networks()` in webasync file, and moved the network scan to `setup()`. It's working well, the options page is now displayed correctly with the list of wifi networks around.

jsleroy, 2022-01-31:
Now that option page is displayed, I notice that if there's networks found, there's one selected in the wifi section. So when I change a parameter, and click apply, it tries to connect to the default selected network. It would be nice to have an option for no wifi network, when using Pedalino in AP mode for example.

alf45tar, 2022-02-02:
Are you using normal boot mode?
I suggest to switch to boot mode 4 or 5: AP with or without Bluetooth.

prostealth, 2022-02-04:
use chrome or edge..
and if you making changes on option and save it, dont refresh it. it will resubmit the command and make the pedalino restarting again. instead close the the browser tab and open 192.168.4.1 again..

alf45tar, 2022-02-05:
I cannot replicate your issue.

jsleroy, 2022-02-05:
I'm using Chrome.
I'm not surprised this can't be replicated.
I still have to try using different boot mode. Or I could also live with networks founds at boot only, as my patch is doing.
BTW, how do change the boot mode ?

alf45tar, 2022-02-05:
To select a boot mode:

- Press and release EN button and immediately after press and hold BOOT button

- Follow the istructions on display. Keep it pressed until the progress bar reach the end to reset to factory default. If you release the button before the progress bar reach the end PedalinoMini™ will start in one of the supported boot mode.