alf45tar/PedalinoMini #314: How can I set it up to achieve a setting similar to the LINE6 fbv MK2 expression pedal?

bobvc133, 2022-02-05:
Hello, how can the pedalinomini be set up to achieve a setting similar to the LINE6 fbv MK2 expression pedal? The default is VOLUME function and LED, move the pedal to send CC7, when you move the expression pedal until the bottom momentary switch is triggered, activate the WHA function and LED, turn off the VOLUME function and LED, then move the pedal to send CC11
[image]
[image]

alf45tar, 2022-02-05:
[image]

alf45tar, 2022-02-05:
Pedal 6 is the expression pedal
Pedal 5 is a footswitch
Led 1 is VOLUME
Led 2 is WAH

On pedal 5 press:

- CC 7 is activated/sent
- led 1 is set to green
- led 2 is set to black

On pedal 5 release:

- CC 11 is activated/set
- led 1 is set to black
- led 2 is set to red

Pedal 6 will sent the last CC activated.

bobvc133, 2022-02-05:
Pedal 5 is momentary switch

alf45tar, 2022-02-05:
The only way to send different CC on the same momentary switch is to use 2 banks.

On monetary switch press add 2 actions:

- send CC 7 on first bank or CC11 on second bank
- switch the next bank using Bank+ or Bank-

bobvc133, 2022-02-05:
I tried the second scheme of switching libraries, but some settings cannot be customized, such as LED number and color

alf45tar, 2022-02-05:
For set led color you can use a separate action

closed 2022-02-10 by alf45tar