alf45tar/PedalinoMini #315: Option page reboots device

Designbase, 2022-02-07:
Hi,

I am using a LilyGO TTGO T8 ESP32 connected with WROVER KIT.
It is working fine except as soon as I enter the Option page the device reboots.
Did a fresh install took disconnected Leds and Switches but it is still rebooting.

Any idea where this problem might come from?
Thanks!

alf45tar, 2022-02-07:
Which firmware version did you used? esp32dev or what else?

Designbase, 2022-02-07:
ESP-WROVER-KIT 2.3.2 as shown in the Installer

alf45tar, 2022-02-07:
Did you tried with ttgo-t-eight?

Designbase, 2022-02-07:
No that option does not work it sets the device in a boot loop...

alf45tar, 2022-02-07:
Your board is not directly supporter. Try esp32dev or modify the pins assignment in Pedalino.h.

Designbase, 2022-02-07:
Well the esp32dev did work... Why I used the Wrover is beyond me. Thought I read it somewhere....
Thanks again for your help!

Designbase, 2022-02-07:
Mmmmm... still very unstable, random reboots.
Will look into the pin modification... if I can find it...

Designbase, 2022-02-10:
What Board would you advise that has A. Battery connection and B. External Antenna...
and is compatible with the PedalinoMini ?

For now the LilyGO TTGO T8 ESP32 works and is stable build with the Wrover setting, except for the Option page. 

Thanks

mattzzw, 2022-02-20:
Having the same issue using the DOIT ESP32 board (`esp32doit-devkit-v1`). 
Reboot happens every time I try to access the options page. It does not happen directly after a factory reset.
I tried it with the provided firmware and with the one I compiled myself.
Log output shows that the watchdog got triggered:
```
E (195047) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (195047) task_wdt:  - async_tcp (CPU 0/1)
E (195047) task_wdt: Tasks currently running:
E (195047) task_wdt: CPU 0: IDLE0
E (195047) task_wdt: CPU 1: IDLE1
E (195047) task_wdt: Aborting.
abort() was called at PC 0x401c3864 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400907a0:0x3ffbffe0 0x40090a1d:0x3ffc0000 0x401c3864:0x3ffc0020 0x4008ed29:0x3ffc0040 0x401fea27:0x3ffbc2c0 0x401c58bb:0x3ffbc2e0 0x400931ed:0x3ffbc300 0x40091a2e:0x3ffbc320
```

mattzzw, 2022-02-26:
Hm, the watchdog timeout is not exactly reproducible in 2.3.2 or 2.4.0.
Sometimes(tm) it works. Dunno if it has to do with the number of wifi networks to scan or something odd like that. But in general it reboots like 90% of the time for me when I try to access the options tab.

alf45tar, 2022-02-26:
How many wifi network are reacheable? Can you try in a different location where there are few wifi network?

alf45tar, 2022-02-27:
Just released version 2.4.1. It is a try to fix Options menu reboot in case of too much wifi networks.
It works for me like the previous version because I cannot simulate the real scenario.
Can you @mattzzw test and confirm if it works for you too?

alf45tar, 2022-02-27:
I did a small change. I removed the while loop. A manual refresh of Options page is requested when scanning is finished.

mattzzw, 2022-02-27:
Hm, I erased the ESP, flashed 2.4.1 with your latest commit and it seemed to work fine at first. The options page was reliably accessible.

Then I played around with changing configurations, uploaded my configuration and saved it as current profile and the options page did not load again. Just a part of the menu bar was shown.

[image]

(So that for me points away from the number of wifi networks.)
Log output shows 
```
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
```

alf45tar, 2022-02-28:
`ack timeout 4` is not directly linked to Options page.
It is common to a lot of ESP32 projects and I never find a definitive solution like many others.

closed 2022-03-01 by alf45tar