alf45tar/PedalinoMini #323: Depend on the PlatformIO libraries from a registry

ivankravets, 2022-03-23:
Hi @alf45tar,

Thank you so much for using PlatformIO! 🙏 

Some user (the `Sergey`) has reported a bad developer experience using PlatformIO and this project. See the review https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide&ssr=false#review-details

Could we ask you to use the Semantic Versioning declaration for `lib_deps` and protect your project from breaking changes? Currently, the project pulls THE LATEST code from each Git repository and does not control API-breaking changes. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#version-requirements
Also, most Windows users will have a bad developer experience if they didn't install Git client.

We recommend converting Git dependencies to the [PlatformIO Registry](https://registry.platformio.org) dependencies. Please search for each library and visit the "Installation" tab. For example, 

**current code**
```
https://registry.platformio.org
```ini
lib_deps =
   https://github.com/bxparks/AceButton
   https://github.com/bblanchon/ArduinoJson
```

**updated code``
```
https://registry.platformio.org
```ini
lib_deps =
   bxparks/AceButton@^1.9.2
   bblanchon/ArduinoJson@^6.19.3
```

Thanks in advance!

alf45tar, 2022-03-24:
Thanks @ivankravets. Good to know but PedalinoMini doesn't compile with the latest version of all libraries because for some libraries the latest version is too old.

ivankravets, 2022-03-24:
Could you post here uncompatible libraries?

alf45tar, 2022-03-24:
For example the latest version of [ESPAsyncWebServer](https://registry.platformio.org/libraries/me-no-dev/ESP%20Async%20WebServer) is 2 years old but the latest changes are only 13 hours ago.

However the error is linked to program size. The firmware does not fit anymore into the partition with the latest available versions of the libraries.

> Error: The program size (2074374 bytes) is greater than maximum allowed (1900544 bytes)
>RAM:   [==        ]  21.6% (used 70728 bytes from 327680 bytes)
>Flash: [==========]  109.1% (used 2074374 bytes from 1900544 bytes)

ivankravets, 2022-03-25:
It seems that some library causes this issue. Could you try one-by-one switching to the stable versions and find out which library makes this issue?

alf45tar, 2023-02-13:
Fixed in version 3.0.10.

closed 2023-02-13 by alf45tar