alf45tar/PedalinoMini #324: Physical wiring diagram of PedalinoMini-Case-1

bobvc133, 2022-04-01:
Hello, can you post the physical wiring diagram of the PedalinoMini-Case-1? I would like to refer to the installation wiring

alf45tar, 2022-04-02:
Schematic is published [here](https://github.com/alf45tar/PedalinoMini-Case-1).
What do you mean with physical wiring? A photo of my implementation?

bobvc133, 2022-04-03:
> 

yes i would like to see how you routed the installation as i am considering whether to make a pcb board to avoid messy wiring

bobvc133, 2022-04-03:
[image]

bobvc133, 2022-04-03:
In addition, is the din and dout of the WS2812B in the schematic diagram reversed?

alf45tar, 2022-04-03:
Your PCB cannot works with PedalinoMini-Case1. Components are in a different place and also leds and footswitches are on different plane/level.

alf45tar, 2022-04-03:
> In addition, is the din and dout of the WS2812B in the schematic diagram reversed?

You are right. Schematic is wrong. I will fix it. Thanks.

bobvc133, 2022-04-03:
3D printing the case is a bit expensive, so I made the case out of black acrylic, the cost is only about $5, I am in China, other accessories are very cheap, and the 10-pedal pedal cost only about $30

y0m1g, 2022-04-05:
I would also appreciate a photo of your physical implementation inside the case 🙏

alf45tar, 2022-04-05:
[image]

alf45tar, 2022-04-05:
> > In addition, is the din and dout of the WS2812B in the schematic diagram reversed?
> 
> You are right. Schematic is wrong. I will fix it. Thanks.

Schematic has been fixed

alf45tar, 2022-04-06:
In version 2.4.0 released last February it is possible to link each button to a led in Pedals setup in order to streamline Actions definition.

In other words assign the led to a pedal/button in Pedals setup and use Default value as led in Actions definition.

bobvc133, 2022-04-06:
> In version 2.4.0 released last February it is possible to link each button to a led in Pedals setup in order to streamline Actions definition.
> 
> In other words assign the led to a pedal/button in Pedals setup and use Default value as led in Actions definition.
> 
> The led wiring can be simplified and the order is independent from pedals.

I mean, other than the person who made the PedalinoMini, it may take a few more tests when editing the LEDs to get the order right

closed 2022-04-14 by alf45tar