alf45tar/PedalinoMini #325: TTGO T-DISPLAY USB interface problem

bobvc133, 2022-04-01:
#define USB_MIDI_IN_PIN GPIO_NUM_21 // SDA
#define USB_MIDI_OUT_PIN GPIO_NUM_22 // SCL
#define DIN_MIDI_IN_PIN GPIO_NUM_21
#define DIN_MIDI_OUT_PIN GPIO_NUM_22
TTGO T- DISPALY Can I define both USB and DIN to use GPIO like this? I would like to use a 2-way toggle switch to select whether to use the USB or DIN interface. Also, have you considered using a multiplexer?
#define USB_MIDI_IN_PIN GPIO_NUM_21 // SDA
#define USB_MIDI_OUT_PIN GPIO_NUM_22 // SCL
#define DIN_MIDI_IN_PIN GPIO_NUM_21
#define DIN_MIDI_OUT_PIN GPIO_NUM_22
TTGO T- DISPALY Can I define both USB and DIN to use GPIO like this? I would like to use a 2-way toggle switch to select whether to use the USB or DIN interface. Also, have you considered using a multiplexer?

alf45tar, 2022-04-02:
You do not need to define DIN_MIDI_* as USB_MIDI_*. 
Connect the serial port pins to the 2-way toggle switch it is enough.

If you define USB and DIN on the same pins disable one port otherwise PedalinoMini will send every message twice.

Multiplexer do not makes sense for me because you will save only 1 pin.

bobvc133, 2022-04-03:
There is no need to redefine GPIO, directly connect the serial port of TTGO to the 2-way toggle switch, and then connect the DIN interface and the serial port of ARDUINO_PRO_MICRO to the 2-way toggle switch respectively. You can directly select the DIN or USB interface through the 2-way toggle switch. Look at the schematic I drew, right?
[image]

alf45tar, 2022-04-03:
It should work.

bobvc133, 2022-04-03:
ok i'll try it out on a breadboard when i have time

bobvc133, 2022-04-03:
> It should work.

Are you using TTGO to connect ARDUINO_PRO_MICRO? I tested that the TX1 and RX1 of ARDUINO_PRO_MICRO need to be reversed with the TX1 and RX1 of TTGO before they can be used.

bobvc133, 2022-04-03:
> It should work.

I found that it is not a problem of reverse connection. It seems that the schematic RX and TX and the physical markings are reversed. When I connected, I looked at the markings on the circuit board.
[image]

bobvc133, 2022-04-03:
The interface of the schematic ARDUINO_PRO_MICRO I drew before is reversed, and the following is correct
[image]