alf45tar/PedalinoMini #328: FLIP_SCREEN

bobvc133, 2022-04-14:
#define FLIP_SCREEN  A value of 0 or 1 doesn't seem to invert the screen vertically

alf45tar, 2022-04-15:
It works only for OLED but I am adding to TFT too.

alf45tar, 2022-04-15:
Added from version 2.4.3

bobvc133, 2022-04-15:
> Added from version 2.4.3

I am using an SSD1306 0.96 screen and a SSD1309 compatible SSD1306 2.42 screen. After updating to 2.4.3, the screen still does not vertically invert the screen. Besides modifying the #define FLIP_SCREEN value to 1, which code needs to be modified?

alf45tar, 2022-04-15:
I re-tested and it works for me.
Did you customised the source?

bobvc133, 2022-04-16:
> I re-tested and it works for me. Did you customised the source?

I have several TTGO T-DISPLAY and ESP32 development boards. The TTGO can flip the screen vertically, but the ESP32+SSD1306 screen can't. I want to know what code needs to be modified?

alf45tar, 2022-04-16:
https://github.com/alf45tar/PedalinoMini/blob/f265fe8a7ad44d1fbcbff3457a15359d05c61882/src/DisplayOLED.h#L1147

bobvc133, 2022-04-16:
> https://github.com/alf45tar/PedalinoMini/blob/f265fe8a7ad44d1fbcbff3457a15359d05c61882/src/DisplayOLED.h#L1147
Sponsored version DisplayOLED.h code is different

alf45tar, 2022-04-16:
You are right. My fault.
Fixed now.

bobvc133, 2022-04-16:
OK problem solved, #define FLIP_SCREEN value is 0 when the screen is reversed, 1 is normal

closed 2022-04-16 by alf45tar