alf45tar/PedalinoMini #330: MIDI pedal finished product release & LED color issue

bobvc133, 2022-04-26:
This is my pedal! The latest sponsored version firmware has a problem with the LED color setting, it is set to red, but the actual display is green.
[image]
[image]
[image]
[image]
[image]
[image]

mknerr, 2022-04-26:
That's beautiful! 
Regarding your LED issue, I can't say for certain, but having worked with pixel strip before, I know the colors are not always addressed in the same order (ie., GRB, RGB, etc.), so you probably just need to tweak the code somewhere to adjust that. 
If you try to make them Red, do they turn green?
@alf45tar, Perhaps an option could be added to the settings to make it easier to adjust. Probably not a big deal; you'd really only need to do this once.

bobvc133, 2022-04-26:
Yes, if you often need to modify the code to customize the settings, then it is meaningless to update the firmware through the browser, backup and restore the configuration file and other regular operations. Ordinary users basically cannot upgrade the firmware unless the firmware is a customized version. It is recommended to add custom settings such as the number of LEDs, vertically flip the screen, etc. to the WEB interface.

mknerr, 2022-04-26:
Yeah, good point.

alf45tar, 2022-04-26:
@bobvc133 it is amazing

alf45tar, 2022-04-26:
> That's beautiful!
> Regarding your LED issue, I can't say for certain, but having worked with pixel strip before, I know the colors are not always addressed in the same order (ie., GRB, RGB, etc.), so [...]

Did you try to change the following setting?

https://github.com/alf45tar/PedalinoMini/blob/93f915d08a022515a33ca77fab4e160ca2a61f90/src/Pedalino.h#L28

bobvc133, 2022-04-27:
> Did you try to change the following setting?
>
> https://github.com/alf45tar/PedalinoMini/blob/93f915d08a022515a33ca77fab4e160ca2a61f90/src/Pedalino.h#L28

You can customize the settings by modifying the code, but I have to re-flash the machine. It is basically impossible to update the firmware through the browser. VS and PlatformIO sometimes have more problems. I hope that these custom items can be set in the WEB interface, and the flashing machine is simple. Just upgrade the firmware

alf45tar, 2022-04-27:
Thanks @bobvc133 for the suggestion. I will try to add as much as possible every configuration parameters in the web interface with the next major release.
Another planned improvement is to remove use of EEPROM and use only configuration files to avoid problem on version upgrade.

bobvc133, 2022-04-27:
> Thanks @bobvc133 for the suggestion. I will try to add as much as possible every configuration parameters in the web interface with the next major release. Another planned improvement is to remove use of EEPROM and use only configuration files to avoid problem on version upgrade.

Looking forward to the new version of the firmware, thank you for your efforts!

closed 2023-01-07 by bobvc133