alf45tar/PedalinoMini #333: Jog Wheel not working

bobvc133, 2022-05-03:
The Jog Wheel doesn't seem to be working properly，ttgo t-display 2.4.4

bobvc133, 2022-05-07:
[image]
I added two more KY-040s to the pedals, the switch and the resistance ladder are connected, and the Jog Wheel is connected to A3, D3, A4, D4, they are all built according to the schematic diagram, but they can't work properly

alf45tar, 2022-05-07:
For the moment you can use a workaround to fix the jog wheel: add at least one action to bank 0.

bobvc133, 2022-05-09:
> For the moment you can use a workaround to fix the jog wheel: add at least one action to bank 0.

I added easing in BANK 0, KY-040 still doesn't work

closed 2022-10-15 by alf45tar