alf45tar/PedalinoMini #335: [Questions] Relating to the Heltec board

Samantha-uk, 2022-05-06:
Firstly I must say this is an **amazing** work of art!

Like others (_I think_) I did not find myself in a position to purchase a retail Midi Footswitch.  Having some experience in s/w & h/w I set out to design/build one myself and in my research found your gem!

I'm working with a Heltec WiFi 32 and have been able to locally build and flash the image using the VS platform.io extension.

At this point I'm building an 8 button box.

**GPIOs**
I see that not all of the available GPIOs are used, is this because some of the other ESP32 boards do not present as many pins?

**Momentary 2**
Is it possible to modify the code so that when I set a Pedal to Momentary 2, it will set both the inputs to digital and use internal pull-ups?  If yes, could you point me at the area in the codebase to look at (_It's been many years since I used C/C++ in anger and I'm new to MC dev_) and I can have a play.

**Simultaneous button press**
I see you use the AceButton library, which says it can support simultaneous button press detection [example](https://github.com/bxparks/AceButton/blob/develop/examples/SimultaneousButtons/SimultaneousButtons.ino).  Do you think it is feasible to incorporate this feature? (_I am happy to try to submit a PR but don't want to pursue this if there are known reasons it will not work_)

alf45tar, 2022-05-06:
Pin assignment start here
https://github.com/alf45tar/PedalinoMini/blob/86ef84bb3b876c670c29822e3a7f52a044912536/src/Pedalino.h#L42
and I do no see any unused pin according the following [schematic](https://resource.heltec.cn/download/WiFi_Kit_32/WIFI_Kit_32_pinoutDiagram_V2.pdf).

alf45tar, 2022-05-06:
GPIOs 34 to 39 are input only pins. These pins don’t have internal pull-up or pull-down resistors.

alf45tar, 2022-05-06:
Is simultaneous button press important for you?
It is feasible but it not so easy and also...

 >Note that this program works only when using ButtonConfig. Buttons using
 >EncodedButtonConfig (binary encoding) and LadderButtonConfig (resistor
 >ladder) cannot detect simultaneous button presses.

Samantha-uk, 2022-05-07:
After much digging I've discovered that my HELTEC_WIFI_KIT_32 is using a ESP32-D0WDQ6 soc.
This appears to be the Initial production release chip of the ESP32 series and is known to be a bit buggy.
Thankfully no soldering was done so I can return it.

I'm going to proceed with a [Unexpected Maker ProS3](https://esp32s3.com/index.html#home) , it costs a little bit more (_But given I need just the one it is not significant_) but it uses the latest gen ESP32-S3 soc and the board is amazingly powerful and well constructed.

I can see two downsides.
1) No integrated display
2) Arduino core support is only a few days old!

[image]

[image]

closed 2022-05-07 by Samantha-uk