alf45tar/PedalinoMini #336: [Issue] [E][esp32-hal-adc.c:135] __analogRead(): GPIO13: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.

Samantha-uk, 2022-05-06:
Using firmware  2.4.4 on a heltec_wifi_kit_32.

When I open the "Logs & Console" view, I see

[E][esp32-hal-adc.c:135] __analogRead(): GPIO13: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.

being printed about once a second.

[image]

alf45tar, 2022-05-06:
I do not own an heltec_wifi_kit_32 to test but it looks like it is an issue of the Heltec board.

Heltec is using GPIO13 (connected to ADC2) as analog pin to read the battery voltage but since the ADC2 module is also used by the Wi-Fi, only one of them could get the preemption when using together, which means the adc2_get_raw() may get blocked until Wi-Fi stops, and vice versa.

More information can be found here

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html

or searching "ESP32 ADC2 wifi conflict" on google.

Samantha-uk, 2022-05-06:
I shall have to have a dig :smile:

Samantha-uk, 2022-05-07:
You're absolutely right.
I'm giving up on the Heltec.

closed 2022-05-07 by Samantha-uk