alf45tar/PedalinoMini #338: Can I directly burn the edited firmware.bin or spiffs.bin through flash_download_tool?

bobvc133, 2022-05-14:
Can I directly burn the edited firmware.bin or spiffs.bin through flash_download_tool? How to get the burning address? If possible, I can easily erase the flash to burn the off-the-shelf firmware when there is no VC or when there is a problem with PlatformIO, because I have to log in to GITHUB through VPN, use VC and WEB interface to get updates, because of network reasons cause various problems Frequent
[image]

alf45tar, 2022-05-14:
Never use the tool but the addresses are here https://github.com/alf45tar/PedalinoMini/blob/master/src/partitions_ota.csv

bobvc133, 2022-05-14:
All kinds of problems often occur when connecting through a VPN, so I want to find a more convenient way to meet my daily needs