alf45tar/PedalinoMini #339: Modify default settings

bobvc133, 2022-05-23:
Where can I modify the default resistance ladder value and pedal configuration above 6? Inexplicable problems often occur when modifying the default pedals in the WEB interface

alf45tar, 2022-05-23:
I am not sure to understand your question.
Having more than 6 buttons is not reliable. The voltage separation between them is not enough to guarantee a reliable detection and avoid spurious press.
You can use any resistors in your ladder and PedalinoMini can recognize it via ladder calibration.
Correct resistors calculation is mandatory to have enough voltage separation.

bobvc133, 2022-05-24:
The question of translation, what I mean is to modify which code of which file can I directly compile the WIFI username and password, resistance ladder value and pedal settings into the firmware? In this way, I do not need to set these items when I update or restore the factory settings. In addition, there are often inexplicable problems when setting pedal 7 and pedal 8 in the WEB interface.

alf45tar, 2022-05-24:
Default values are here

https://github.com/alf45tar/PedalinoMini/blob/760e8ef91ea89b79bfd6b8792047e21ff303e41b/src/Config.h#L721

closed 2022-05-24 by bobvc133