alf45tar/PedalinoMini #341: ESP Web Tools

bobvc133, 2022-05-26:
Because I am in China, I need a VPN to use ESP Web Tools, so I configured an ESP Web Tools server test, but found a problem, my TTGO T-DISPLAY memory is 16M, built by VS or you ESP Web Tools burns the firmware, the WEB interface can display the normal size of 16m, but the memory of the ESP Web Tools I built only shows 4M after burning, have you encountered such a problem? How is it solved?

alf45tar, 2022-05-26:
ESP32 has the ability to also use up to 4 MB of external SPI RAM memory. The external memory is incorporated in the memory map and, with certain restrictions, is usable in the same way as internal data RAM.

https://thingpulse.com/esp32-how-to-use-psram/
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html

bobvc133, 2022-05-26:
[image]
[image]
[image]
[image]
The first two pictures are the results of upgrading the firmware on the ESP Web Tools page I built, and the last two pictures are the results of upgrading the firmware on the ESP Web Tools page you built. The hardware and firmware used are the same, but the Flash Size is different.

alf45tar, 2022-05-27:
Did you changed partitions file?

bobvc133, 2022-05-27:
> Did you changed the partitions file?
> 
> https://github.com/alf45tar/PedalinoMini/blob/master/src/partitions_ota.csv

I just modified the firmware link address of ttgo-t-display-2.4.0.json, because I installed ESP Web Tools on my server and copied the 2.4.0 version firmware you released, so I didn't modify other files at all, the only one The difference is that I use the ESP Web Tools installed on my own server to upgrade the firmware. Is it because my ESP Web Tools configuration is wrong that the Flash Size is different?

alf45tar, 2022-05-28:
The SPI flash size is configured by writing a field in the software bootloader image header, flashed at offset 0x1000.

By default, the SPI flash size is detected by esptool.py when this bootloader is written to flash, and the header is updated with the correct size. Alternatively, it is possible to generate a fixed flash size by setting CONFIG_ESPTOOLPY_FLASHSIZE in the project configuration.

bobvc133, 2022-05-28:
> The SPI flash size is configured by writing a field in the software bootloader image header, flashed at offset 0x1000.
> 
> By default, the SPI flash size is detected by esptool.py when this bootloader is written to flash, and the header is updated with the correct size.
> 
> esptool.py is used by VSC but not by ESP Web Tool.

Maybe you don't understand what I mean. Below the screenshot above, I have explained that when I use the ESP Web Tool page you built to upgrade the firmware, it will correctly identify the flash size, the Flash Size is 16m, but after the ESP Web Tool page I built is upgraded However, it shows that the Flash Size 4m, ttgo-t-display-2.4.0.json and other files have not been modified, and the path is "https://raw.githubusercontent.com/alf45tar/PedalinoMini/2.4.0/firmware/ttgo- t-display/”. VSC is not used.
[image]

closed 2022-06-20 by bobvc133