alf45tar/PedalinoMini #342: I made one

Samantha-uk, 2022-06-01:
I've just finished making one of these :)

It is based on an AZ Delivery WROOM-32 Dev Kit v4 board, which I added to my platformio.ini
```
[platformio]
default_envs = 
	esp32dev
	esp32doit-devkit-v1
	esp32doit-devkit-v1-ble
	esp32doit-devkit-v1-wifi
	esp-wrover-kit
	heltec_wifi_kit_32
	ttgo-t-eight
	ttgo-t-eight-ble
	ttgo-t-eight-wifi
	ttgo-t-display
	az-delivery-devkit-v4

...

[common]
build_flags_az-delivery-devkit-v4 = 
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D AZ_DELIVERY_DEVKIT_V4
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
;	-D BATTERY
	-D SSD1306WIRE
	-D WEBSOCKET
	-D DIAGNOSTIC
...

[env:az-delivery-devkit-v4]
build_flags = ${common.build_flags_az-delivery-devkit-v4}
upload_port = /dev/tty.usbserial-0001
monitor_port = ${env:az-delivery-devkit-v4.upload_port}
```

[image]

the only issue I've run into is that to flash the firmware or attach to the monitor, one needs to hold the boot button on the wroom board to tell it to listen.  I read somewhere that by soldering a small cap in place this can be avoided, not tried it yet.

I found I could not initially build the firmware, seeing the following error:
```
Compiling .pio/build/az-delivery-devkit-v4/libb64/NimBLE-Arduino/nimble/nimble/host/src/ble_hs_resolv.c.o
.pio/libdeps/az-delivery-devkit-v4/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_cmd.c:31:26: fatal error: soc/soc_caps.h: No such file or directory
compilation terminated.
*** [.pio/build/az-delivery-devkit-v4/libb64/NimBLE-Arduino/nimble/nimble/host/src/ble_hs_hci_cmd.c.o] Error 1
```

I found that pinning the NimBLE-Arduino library to version 1.3.7 resolved this.
I changed
https://github.com/alf45tar/PedalinoMini/blob/master/platformio.ini#L141
to
```https://github.com/h2zero/NimBLE-Arduino/archive/refs/tags/1.3.7.zip```

I decided I wanted 8 'digital' switches (_and did not want to use pull down resistors_) so I modified the pedal configuration for this board; I don't intend to use analog so I 'stole' GPIO_NUM_32 & GPIO_NUM_33
```
#elif defined AZ_DELIVERY_DEVKIT_V4
#undef PEDALS
#define PEDALS 8
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_14, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_32, GPIO_NUM_33};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_39, GPIO_NUM_34, GPIO_NUM_35, GPIO_NUM_0, GPIO_NUM_0, GPIO_NUM_0, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN GPIO_NUM_0
#define USB_MIDI_IN_PIN GPIO_NUM_18
#define USB_MIDI_OUT_PIN GPIO_NUM_19
#define DIN_MIDI_IN_PIN GPIO_NUM_15
#define DIN_MIDI_OUT_PIN GPIO_NUM_4
#define BATTERY_PIN GPIO_NUM_36 // TODO: Check into this
#define FASTLEDS_DATA_PIN GPIO_NUM_5
#else
```

Seems to work OK.
```
Pedal: 1     Button: 1    EventType: 0     ButtonState: 0
Action: PRESET -
PROGRAM CHANGE.....Program   0.....Channel  4
Received from BLE MIDI  ProgramChange 0x00   Channel 04
Pedal: 1     Button: 1    EventType: 1     ButtonState: 1
Pedal: 2     Button: 1    EventType: 0     ButtonState: 0
Action: SNAP 1
CONTROL CHANGE.....Code  69......Value   0.....Channel  4
Pedal: 2     Button: 1    EventType: 1     ButtonState: 1
Pedal: 3     Button: 1    EventType: 0     ButtonState: 0
Action: SNAP 2
CONTROL CHANGE.....Code  69......Value   1.....Channel  4
Pedal: 3     Button: 1    EventType: 1     ButtonState: 1
Pedal: 4     Button: 1    EventType: 0     ButtonState: 0
Action: SNAP 3
CONTROL CHANGE.....Code  69......Value   2.....Channel  4
Pedal: 4     Button: 1    EventType: 1     ButtonState: 1
Pedal: 4     Button: 1    EventType: 2     ButtonState: 1
Pedal: 5     Button: 1    EventType: 0     ButtonState: 0
Action: PRESET +
PROGRAM CHANGE.....Program   1.....Channel  4
Received from BLE MIDI  ProgramChange 0x01   Channel 04
Pedal: 5     Button: 1    EventType: 1     ButtonState: 1
Pedal: 6     Button: 1    EventType: 0     ButtonState: 0
Action: TUNER
CONTROL CHANGE.....Code  68......Value   0.....Channel  4
Pedal: 6     Button: 1    EventType: 1     ButtonState: 1
Pedal: 7     Button: 1    EventType: 0     ButtonState: 0
Pedal: 7     Button: 1    EventType: 1     ButtonState: 1
Pedal: 8     Button: 1    EventType: 0     ButtonState: 0
Pedal: 8     Button: 1    EventType: 1     ButtonState: 1
```

**But what does it LOOK like!?**
[image]

[image]
[image]
[image]

**And those extra windows!?!?**
I want to have a scribble strip for each footswitch, so I've left space to position an OLED display between pairs of switches.
I'm aiming for something like this
[image]

I aim to use a multiplexer to resolve the I2C addressing issue.
https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/overview
explains how.

I just need to wrap my head around the OLED display code to support this and modify it accordingly; then wire in the multiplexer and additional OLED display panels.

rigr, 2022-06-01:
Congratulations, it looks great. Do you aim to share the stl file for your
box?
Thx! Richard
Message ID: ***@***.***>

>

Samantha-uk, 2022-06-02:
> Congratulations, it looks great. Do you aim to share the stl file for your box? Thx! Richard Message ID: ***@***.***>
> […](#)

Here you go :)

[enclosure.zip](https://github.com/alf45tar/PedalinoMini/files/8819939/enclosure.zip)

mknerr, 2022-06-02:
Very nice! And thanks for sharing your files with the community.
What wireless protocol are you using to connect to the Stomp, and which dongle are you using?
Thanks!
Matthew

Samantha-uk, 2022-06-02:
Matthew,
I've got a CME Widi Master plugged into the Stomp.
[image]
I got it so I could use MIDI Designer app on my iPad.
The Widi master seems quite good.  I can connect to it from Mac OS, Android, Pad OS and, of course, Pedalino.
The system is configured to use BLE MIDI Only; I'm not that knowledgable about it ... I _believe_ it is supported by the NimBLE library.
[image]
Connection is pretty seamless ... whichever device I turn on second, the BLE link seems to establish in just a few seconds.

mknerr, 2022-06-06:
Okay, so can Pedalino connect directly to the widimaster, or only via another device? Thanks,

Samantha-uk, 2022-06-06:
When I am using my setup in "gig" mode I have the following

- My Pedalino Box.
- An HX Stomp with a WIDIMaster plugged into it.

When I power it all on, it connects without further intervention.  The Pedalino box starts up quicker than the Stomp.

The only other things I need are in my sound chain (Guitar/cables/amp/cab) 🎸 🔈 😄

mknerr, 2022-06-06:
Very nice! Thanks for the explanation.

pruttelherrie, 2022-06-27:
> **And those extra windows!?!?** I want to have a scribble strip for each footswitch, so I've left space to position an OLED display between pairs of switches. I'm aiming for something like this [...]

Hi Samantha, you could possibly do without the multiplexer, take a look at this: https://github.com/bitbank2/Multi_OLED

I have done something similar with cheap chinese colour SPI displays and the TFT_eSPI library, but that won't work with Pedalino I think.

DWSQUIRES, 2022-08-25:
Did you ever manage to get the multi displays working? Very interested.
I'm also building one to use with an HX Stomp!

Samantha-uk, 2022-08-26:
I never did as I got sidetracked with another project (patchbox OS based
multi effects, it can run the OSS that Moddevices use on their Dwarf and
other pedals).  I had intended to go down this kind of route (
https://github.com/bitbank2/Multi_OLED).

Sadly all my possessions were destroyed in a home fire a couple of weeks
ago so it’s going to be some time before I can even consider trying it out.

DWSQUIRES, 2022-08-26:
Oh no! That's so sad! Did you loose your instruments and pedals too?

Samantha-uk, 2022-08-27:
Sadly yes, including a bass I’d hand built and the wood/bits for the second
hand build.  I focus on the fact that I survived and no one got hurt!

closed 2024-05-16 by alf45tar