alf45tar/PedalinoMini #347: Action page restart and garbled problem

bobvc133, 2022-06-15:
Editing action events in the action page, deleting and saving sometimes causes restart or garbled problems
[image]

bobvc133, 2022-06-15:
There's also the problem that creating new configs and applying & saving available configs often fails, and only succeeds if you do it right after a reboot.

alf45tar, 2022-06-15:
Are you using the last sponsor version?

alf45tar, 2022-06-15:
Can you attach the log?

bobvc133, 2022-06-17:
> Can you attach the log?

Executing task: C:\Users\dada\.platformio\penv\Scripts\platformio.exe device monitor --environment ttgo-t-display <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM12  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ChipRevision 3, CPU Freq 240 MHz, SDK Version v3.3.5-1-g85c43024c
Flash Size 4194304, Flash Speed 40000000 Hz
Internal Total Heap 260156, Internal Free Heap 231348
PSRAM Total Heap 0, PSRAM Free Heap 0
eFuse Vref:1086 mV
nvs Total 65536
PlatformIO Built Environment: ttgo-t-display
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \  
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/  
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini 

Hostname: 67F23A08
PSRAM not found
Bluetooth Classic disabled: 15248 bytes released     
Internal Total Heap 275492, Internal Free Heap 246500
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      67F23A08 
[NVS][Global][Boot Mode]:        1        
[NVS][Global][STA SSID]:         DADAHOME 
[NVS][Global][STA Password]:     bobvc2013
[NVS][Global][AP SSID]:          Pedalino-67F23A08
[NVS][Global][AP Password]:      67F23A08
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    67F23A08
[NVS][Global][Bootstrap Theme]:  bootstrap        
[NVS][Global][Current Profile]:  1  
[NVS][Global][Tap Dance Mode]:   0  
[NVS][Global][Bank Switch]:      0  
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5  
[NVS][Global][LedsOnBright]:     5  
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Ladder]:           Ladder 1 Level 0  
[NVS][Global][Ladder]:           Ladder 2 Level 140
[NVS][Global][Ladder]:           Ladder 3 Level 338
[NVS][Global][Ladder]:           Ladder 4 Level 520 
[NVS][Global][Ladder]:           Ladder 5 Level 751 
[NVS][Global][Ladder]:           Ladder 6 Level 1010

Boot NORMAL
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
Connecting to
SSID        : DADAHOME
Password    : bobvc2013
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START 
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 203 - ASSOC_FAIL
SYSTEM_EVENT_STA_DISCONNECTED
SYSTEM_EVENT_STA_CONNECTED     
[... 247 lines cut ...]
/blueboard.cfg
/mustang.cfg
/Default.cfg
done.
Looking for configuration files on SPIFFS root ...
/case1.cfg
/blueboard.cfg
/mustang.cfg
/Default.cfg
done.
MTC None
Looking for configuration files on SPIFFS root ...
/case1.cfg
/blueboard.cfg
/mustang.cfg
/Default.cfg
done.
Looking for configuration files on SPIFFS root ...
/case1.cfg
/blueboard.cfg
/mustang.cfg
/Default.cfg
done.

This is the log of applying & saving the available configuration after restarting the device. First of all, applying & saving the available configuration to group B succeeds, but applying & saving the available configuration to group C fails. There is nothing in the action page, and occasionally it still works will restart strangely

bobvc133, 2022-06-17:
Executing task: C:\Users\dada\.platformio\penv\Scripts\platformio.exe device monitor --environment ttgo-t-display <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM12  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
BLE MIDI service advertising started
Connecting to
SSID        : DADAHOME
Password    : bobvc2013
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START 
SYSTEM_EVENT_STA_CONNECTED     
BSSID       : 90:17:C8:B1:95:D0
RSSI        : -58 dBm
Channel     : 1
STA         : 08:3A:F2:67:83:DC
SYSTEM_EVENT_STA_GOT_IP        
Hostname    : 67F23A08
IP address  : 192.168.3.12
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.3.1
DNS 1       : 192.168.3.1
DNS 2       : 0.0.0.0
mDNS responder started
Internal Total Heap 271996, Internal Free Heap 146332
Loading profile ...
Reading NVS Profile C ... OTA update started
HTTP server started
Connect to http://67F23A08.local/update for firmware update
Connect to http://67F23A08.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
done
Bank  2
Pedal  1     1   LADDER       PRESS_1       POLARITY+      Channel  0   Pin D25   Pin A36
Pedal  2     1   LADDER       PRESS_1       POLARITY+      Channel  0   Pin D26   Pin A37
Pedal  3     None   ANALOG                     POLARITY-      Channel  0   Pin A38 D27   
Pedal  4     None   ANALOG                     POLARITY+      Channel  0   Pin A39 D12
Pedal  5     None                              POLARITY+      Channel  0
Pedal  6     None                              POLARITY+      Channel  0
Pedal  7     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D35
Pedal  8     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D0 
Pedal  3 calibration max 1023
Pedal  3   input 1023 output 1023 velocity 0.16
Pedal  4 calibration max 1023
Pedal  4   input 1023 output 1023 velocity 0.16
MTC None
Event: 1
Loading profile ...       
Reading NVS Profile A ... done
Bank  2
Pedal  1     1   LADDER       PRESS_1       POLARITY+   CONTROL_CHANGE       0   Channel  1   Pin D25   Pin A36
Pedal  2     1   LADDER       PRESS_1       POLARITY+   PROGRAM_CHANGE       0   Channel  1   Pin D26   Pin A37
Pedal  3     None   ANALOG                     POLARITY-   CONTROL_CHANGE      11   Channel  1   Pin A38 D27   
Pedal  4     None   ANALOG                     POLARITY+   CONTROL_CHANGE      12   Channel  1   Pin A39 D12
Pedal  5     None                              POLARITY+      Channel  0
Pedal  6     None                              POLARITY+      Channel  0
Pedal  7     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D35
Pedal  8     12L   MOMENTARY1   PRESS_1_2_L   POLARITY+      Channel  0   Pin D0 
Pedal  3 calibration max 1023
Pedal  3   input 1023 output 1023 velocity 0.05
Pedal  4 calibration max 1023
Pedal  4   input 1023 output 1023 velocity 0.05
MTC None
abort() was called at PC 0x401ead4f on core 1

ELF file SHA256: 0000000000000000

[... 42 lines cut ...]
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Ladder]:           Ladder 1 Level 0
[NVS][Global][Ladder]:           Ladder 2 Level 140
[NVS][Global][Ladder]:           Ladder 3 Level 338
[NVS][Global][Ladder]:           Ladder 4 Level 520
[NVS][Global][Ladder]:           Ladder 5 Level 751
[NVS][Global][Ladder]:           Ladder 6 Level 1010

Boot NORMAL
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
Connecting to
SSID        : DADAHOME
Password    : bobvc2013
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 3 - AUTH_LEAVE
SYSTEM_EVENT_STA_DISCONNECTED
WiFi provisioning started

This is the log of the restart after deleting some events after copying the library

alf45tar, 2022-06-24:
Solved some memory fragmentation issue with version 2.4.7. 
Please give a try.

bobvc133, 2022-06-25:
> Solved some memory fragmentation issues with version 2.4.7. Please give a try.

ok！

bobvc133, 2022-06-25:
> Solved some memory fragmentation issues with version 2.4.7. Please give a try.

The only difference between 2.4.7 and 2.4.6 I see is the version number. I have updated the 2.4.6 version, and I have not tested the problem of garbled characters and restarts.
[image]

bobvc133, 2022-06-25:
> Solved some memory fragmentation issues with version 2.4.7. Please give a try.

The position of the IF statement added to the navigation menu is different, will it have any effect?
[image]

alf45tar, 2022-06-25:
The position of the IF statement is not important. The goal is to keep the page string less than 4k.

bobvc133, 2022-06-25:
> The position of the IF statement is not so important. The goal is to keep the page string less than 4k.

Ok, I will update and test it, because I have modified a lot of code, including the translation of the web interface. If the code is updated a lot, I need to re-edit and check for errors, which is very time-consuming each time.

bobvc133, 2022-06-26:
> The position of the IF statement is not so important. The goal is to keep the page string less than 4k.

The test found that garbled characters appeared on the 2.4.7 configuration page
[image]
[image]

alf45tar, 2022-06-26:
I know. It is fixed in 2.4.8 release just now.

bobvc133, 2022-06-26:
> I know. It is fixed in 2.4.8 release just now.
OK, I'll continue testing tomorrow

bobvc133, 2022-06-27:
> I know. It is fixed in 2.4.8 release just now.

"Apply and save" the available configuration to the current configuration file, and occasionally restart when the current library configuration is not empty

bobvc133, 2022-06-27:
> I know. It is fixed in 2.4.8 release just now.

Another problem is that the screen is flipped vertically. TTGO T-DISPLAY is normal. Development boards such as esp32doit-devkit-v1 are flipped by default and need to be flipped again on the options page.

alf45tar, 2022-06-27:
> > I know. It is fixed in 2.4.8 release just now.
> 
> Another problem is that the screen is flipped vertically. TTGO T-DISPLAY is normal. Development boards such as esp32doit-devkit-v1 are flipped by default and need to be flipped again on the options page.

esp32doit-devkit-v1 has no display. By default it is using the default orientation of the OLED library.

bobvc133, 2022-06-28:
> I know. It is fixed in 2.4.8 release just now.

2.4.8 The boot screen seems to be gone

rigr, 2022-06-28:
Installed 2.4.8 yesterday and also had 3 issues:
1) 2 seconds after boot mine provides Access point - no boot-screen.
2) esp32doit-devkit-v1 the screen is flipped after install (with previous
version it was not flipped). It can be re-flipped, when a connection can be
established.
3) the battery Symbol is displayed by default, I commented it out in line
12 of platformio.ini

Apart from that it looks good.

Best regardsMessage ID: ***@***.***
com>

bobvc133, 2022-06-29:
> @rigr
> 
> 1. Boot screen has been disabled to save memory. Uncomment the code in DisplayTFT.h to re-enable it.
> 2. The default value is changed. To flip it go to Options and select Flip Screen.
> 3. You are right. It can be disabled in platformio.io. May be it is better to disable by default.

When the 2.4.8 firmware is turned on and connected to the network, click to skip SmartConfig mode, and the device will restart

rigr, 2022-06-29:
Reinstalled now and get no access to network. :( And pedalino does not provide access point, it only shows "Use the mobile app to smartconfig" and "Provisioning WIFI" and switches between those after a minute each.

rigr, 2022-06-29:
Ok, success to get accesspoint mode: after start immediately press pedal7
and release in the right moment. IT's a bitte too fast, 8 seconds between
the steps would be better. Need to find that setting in the code.

bobvc133, 2022-06-29:
> Ok, success to get accesspoint mode: after start immediately press pedal7 and release in the right moment. IT's a bitte too fast, 8 seconds between the steps would be better. Need to find that setting in the code. bobvc133 ***@***.***> schrieb am Mi., 29. Juni 2022, 06:49:
> […](#)

Not the same thing, one is the boot mode selection, the other is the networking method selection

rigr, 2022-07-07:
Thank you for asking, Alf45tar.
The boot-problem seems to be solved, I can get access to the web interface by selecting the AccessPoint mode.
But there are still some issues -
1) I can not save a configuration and reload it - when I try, a configuration called "case1"(that never existed) is loaded and it can not be edited with the built in configurations editor.
2) I tried several android apps to send wifi credentials but had no success to establish a connection and connect the pedalino mini to my network. I can however get access and set the ssid and password over the webinterface. What app should be used?
3) I have some random crashes when making settings in the webinterface.

I LOVE the neopixel animation you implemented for the different modes. :)

BTW: I installed pedalino mini on a TTGO board with a small OLED and a battery compartment and had to assign  the pins for the display.  It works but I will rather not use that board, because it's good for experiments but too big and the display is on a bad place.

Thank you!
Richard

alf45tar, 2022-07-07:
If case1 is not present is because you didn't updated the system image recently.
Click “Upload File System Image” under Platform.

I suggest also to reset PedalinoMini to factory default as explained here https://github.com/alf45tar/PedalinoMini#booting-modes

The easy way to provide wifi credentials is to follow the detailed instructions here  https://github.com/alf45tar/PedalinoMini#how-to-upload-firmware-wifi-provisioning-and-access-to-web-user-interface.

To solve most of the crashes use the latest version. There is a big improvement on memory fragmentation that it is the root cause of most reboot.

rigr, 2022-07-12:
Oh, I should have mentioned that I had those issues using 2.4.9.
Is there a way to ONLY upload the wifi credentials using a serial link and the http://alf45tar.github.io/PedalinoMini/installer?

alf45tar, 2022-07-12:
Do not define SMARTCONFIG or WPS in platformio.ini

closed 2022-12-10 by alf45tar