alf45tar/PedalinoMini #350: [Feature idea] Pressure sensitive foot switch

FelixMuellCode, 2022-06-20:
I did some testing with a simple foot switch which sent a midi note to my ipad which then triggered a drum sound. 
The latency was so amazingly low that you could absolutely do a live performance with it.

My idea is now, to build a pressure sensitive foot sensor using a piezo or a force sensor. The goal is to be able to change the midi intensity value from my foot stomping on the 'drum' pedal.

With the piezo some debouncing would certainly be necessary as the piezo picks up all the vibrations from the stomp on the pedal.
Furthermore, maybe a protection circuit with a  Zener diode and resistor would be necessary.

If you're interested in this idea I would do some tests with the hardware and then get back at you.

alf45tar, 2022-06-21:
Yes, I am. Which hardware are you planning to use?
I suggest to start testing configuring the pedal as Analog or Momentary and check the result.

FelixMuellCode, 2022-06-21:
Awesome, as of right now I'm using a ESP32 dev board.
I want to order some lolin32s together with the piezos.

I will order from China, so it will take a few weeks.
I will update when I have results.

alf45tar, 2022-12-10:
I am closing the issue. Reopen if you have update.

closed 2022-12-10 by alf45tar