alf45tar/PedalinoMini #352: Make a bigger PedalinoMini

bobvc133, 2022-06-21:
I plan to make another larger PedalinoMini, because the TTGO T-DISPALY screen is smaller, so I plan to use the ESP32-DevKitC-32U IPEX antenna development board + 2.42-inch OLED hardware solution. The biggest problem at present is the power supply:
1. Can 18650 be used to supply power directly through vin?
2. What codes need to be modified for OLED battery status display? For example, the real-time display of no battery status and battery power, the 2.4.5 firmware version now always shows the charging effect.
3. What hardware needs to be added for real-time voltage detection, how to connect, and what code needs to be modified?
4. Can the battery be charged while connecting to the USB? If not, what conflict will there be when supplying power at the same time? Is there any way to solve the charging problem? My current plan is to remove the battery and use other methods to charge it.

Forgive me for asking so many questions at once, I have carefully checked the list of questions, but because some of the answers were long ago, the firmware has been updated many times, and the code has changed a lot, so I can't modify the test according to the original method.
[image]

alf45tar, 2022-06-21:
Question 2: What codes need to be modified for OLED battery status display?

Comment

`-D BATTERY`

in platformio.ini

alf45tar, 2022-06-21:
Question 3: What hardware needs to be added for real-time voltage detection, how to connect, and what code needs to be modified?

I suggest to copy the circuit from TTGO T-Display [schematic](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf) or from other ESP32 boards.

The code to modify is:
https://github.com/alf45tar/PedalinoMini/blob/e29f74c73632bb509429b00aed519aae1180f06b/src/Pedalino.h#L52-L53

alf45tar, 2022-06-21:
Question 4: Can the battery be charged while connecting to the USB? If not, what conflict will there be when supplying power at the same time? Is there any way to solve the charging problem?

Yes, battery can be charged when USB is connected and if you implement the appropriate circuit as suggested in answer to question 3.

bobvc133, 2022-06-22:
> Question 4: Can the battery be charged while connecting to the USB? If not, what conflict will there be when supplying power at the same time? Is there any way to solve the charging problem?
>
> Yes, [...]

Find a simpler solution, directly choose the development board with battery and IPEX antenna, all problems are solved, the only problem may be the adaptation of the firmware program to the battery part, such as charging and power status display, voltage detection, etc.
[image]