alf45tar/PedalinoMini #355: How can I send OSC commands?

smadds, 2022-07-13:
Hi - thanks for a great project. Just loaded it up, but struggling a bit.

I'm trying to connect to a Behringer X32 mixer using OSC. From what I understand the OSC control generally needs a path and a value. How can I set this up in PedalinoMini? I can only see Midi commands and not generic OSC option

alf45tar, 2022-07-13:
OSC works as documented here

https://github.com/alf45tar/Pedalino#open-sound-control-osc

smadds, 2022-07-13:
Thanks for getting back to me so quickly.

I still don't understand the thinking behind the OSC implementation. If I read the doc right, it can only send UDP data on the port it choses, listen on the port it choses and has an OSC address schema unique to itself.

Is the idea that other devices are meant to control the Pedalino? I thought it was designed to control other items of equipment. Surely it needs to be able to configure the ports and the schema of the other equipment?

Am I missing something obvious? I want to use OSC instead of Midi as it is bi-directional and I can show true status on the box - e.g. the mute status of a channel.

alf45tar, 2022-07-14:
You are right. PedalinoMini needs improvements and more flexibility on OSC.

alf45tar, 2022-07-31:
Just release version 2.5.0 with some OSC improvements:
- customise the OSC address
- support for integer and float value

smadds, 2022-08-01:
Brilliant, thanks a lot.

Are you planning to create a release so I can use the installer URL? At the moment it's still only showing up to 2.4.0

smadds, 2022-08-01:
lol - I sponsored you just before I saw your response!

smadds, 2022-08-02:
Just been testing this for my use case (controlling a Behringer X32 mixer) and it works brilliantly, so thank you very much for the extra effort you have put in to this side of the code.

I'd love to be able to get status back from the mixer and use that to control an LED, but I'll raise another ticket for that...

closed 2022-08-29 by alf45tar