alf45tar/PedalinoMini #357: Subscribe for OSC status updates

smadds, 2022-08-02:
Definitely an enhancement request, but maybe wider use than just mine?

Having got the Pedalino Mini to send OSC commends to my device over UDP I would love to get back the true status of the item being controlled - that way if it is changed on the device a local LED can be kept in sync.

To do this I need to regularly send a subscribe command to the device (every 10 seconds), so would need to have a button type that repeats sending every {defined number} of seconds.

It would also need to be able to link an LED to an incoming OSC path and set the colour based on the definable values returned.

For the X32 mixer I am controlling this is described well in the protocol document - Page 77 [here](https://drive.google.com/file/d/1Snbwx3m6us6L1qeP1_pD6s8hbJpIpD0a/view)

smadds, 2022-08-28:
Getting on really well with this code and OSC commands - especially with  TTGO board with the lovely colour display.

Would you have any time to look at an auto repeating button type that could send a message every defined period, @alf45tar ?

alf45tar, 2022-08-29:
A pedal can send a message every defined period using REPEAT PRESSED event.

Use a latch pedal and add an action triggered on REPEAT PRESSED.
Set the repeat interval in Options under Repeat Press Time.

alf45tar, 2022-08-29:
Right now it is the only solution without any code change.
You can replace the latch button with a wire between the corresponding digital pin and GND.

smadds, 2022-08-29:
Wire link sounds like a good workaround!

I'm not able to get the /subscribe to work yet, and I suspect it's because the system is always sending the min/max value as an integer on the end of the OSC command. Unfortunately there are quit a lot of commands that don't need this.

The documentation shows this command as
/subscribe ,si /ch/01/mix/on 2

so, taking a string and then an integer (for update rate). The nearest I can get is by setting the OSC command to
/subscribe ,s /ch/01/mix/on

and the min/max to 2, which gives an output of
/subscribe ,s /ch/01/mix/on~,i~~[     2]

Unfortunately this is not acceptable to the X32.

Is there any way of inhibiting sending the integer value at the end?

alf45tar, 2022-08-29:
Not at the moment. In the new version if Min = Max no value will be sent.

smadds, 2022-08-29:
That's great - patience then!

The /subscribe will allow the mixer to send status updates just when the status changes. Is there a way of parsing these incoming messages to change an LED colour?

onkeldittmayer, 2024-09-28:
Did you find a solution? I have the same problem with a x32