alf45tar/PedalinoMini #359: Battery esp32

SrMorrris, 2022-08-19:
Hello.

I am trying to develop my PedalinoMini, but i have found some issues trying to add bateries to the project..

I am using a _18650 battery_ connected to _USB-Charger TP-4056_.
And i have connected one  _step-up regulator MT-3608_.

This MT-3608 gives you 18 volts in pins OUT+/OUT- and moving the screw on the board...you step down the voltage.
I have found MiniPedalino doesn´t inicciate when voltage it´s more than 3,94 volts (pins VIN and GND on the esp32)
The screen suddenly appears a wifi and turns off, and nothing works good..

When i move the MT-3608 screw at minimum (3,94 volts) the screen suddenly appears connecting to wifi and everything its fine..

But the problem its when minipedalino its turned on for ten minutes, suddenly turn off (the screen goes out) and i have to move again the MT-3608 screw for everything getting to work again..

The voltage work range its very thinny i think or i don´t know what i am doing wrong...i am pretty new at esp-32
I am confused because i have thought that i must feed pins Vin and GND on the esp 32 with 5 volts.
And the screen goes to 3v and GND pins

But not working for me..

Designbase, 2022-08-19:
Dank u voor uw bericht!

Wij zijn vanwege vakantie gesloten tot en met 26 Augustus.

U mail wordt gelezen, indien er calamiteiten zijn wordt er contact met u opgenomen.

Met vriendelijke Groet,

Ronald Westland
Designbase

SrMorrris, 2022-08-19:
I understand holydays..
I think my problem its schematics..
Vin--5V
Esp32 can handle 3.84 maximum
I must down voltage..

I beg for more schematics..
I am not able to connect a potentiometer yet..
Can i connect a 10k potentiometer?

By now i can connect and config pedalino but no idea how to select or change screens...
I see blueboard or case1 have different screens..can i have the two types of screens on my config??

Its my second day with pedalino..
I see its amazing but i feel confused about too many things yet..

I will expect to feel less confused on september..
I hope!
Enjoy holydays!

alf45tar, 2022-08-19:
We are not closed for holidays. @Designbase is not affiliated with PedalinoMini.

alf45tar, 2022-08-19:
I think you should start experiments using the USB power.
Which is your board?
Usually the ESP32 board can be powered with 5V in the VIN board. The ESP32 has an on-board regulator that converts the voltage from 5V to 3.3V.

alf45tar, 2022-08-19:
If you want to connect a battery I suggestion to take inspiration from this board.
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf
Step up regulator is unnecessary.

SrMorrris, 2022-08-19:
Ups..hehe sorry i don't understand before..
Yes you are right...i should start connecting via usb..
But pedalino works perfect with the battery (only when it starts.., its not always, on my case..).
My step up mt-3608 allows me to step down voltage and start system (yes its...)

Ok..so my broblem its my board..i have bought on aliexpress 
ESP32DEVKITV1 30pins
I think the problem its my board which doesn´t have voltage-regulator on the VIN pin..
I will try another method to feed my VIN pin.
I think my Vin pin needs 3,3 volts..

alf45tar, 2022-08-19:
Are you sure you are using VIN pin? VIN is opposite to 3.3V pin.

SrMorrris, 2022-08-19:
Yes certainly I am using Vin Pin and GND.the opposite from the 3v and GND pins
Let me do some capture images from my config and make some photos..
This problem its annoying me hehe
I can initiate pedalino and the screen goes to "connecting to.." 
But thanks to my MT-3608 and moving the screw..

(when i move to minimum the screen doesn´t switch on...i think this means down-voltage)
(when i move max.. the screen switch on for one second with a wifi symbol and then goes off..i this means up-voltage)
I need to move the screw really slow for initiate pedalino.
And then everything goes right...i can switch on/off and everything fine (for a certain time...then i need to move the screw again)

SrMorrris, 2022-08-19:
Hi
This its my board
[image]
[image]
a little bit more closer..
[image]

Thanks Alf4star..pedalino its awesome!!
Now i am understanding tags and screens, and connecting pots..

alf45tar, 2022-12-10:
Can we close the issue?

closed 2023-01-10 by alf45tar